j5lien/esphome-idasen-desk-controller #2: desk_height &  binary_sensor not being updated

IhorSyerkov, 2021-02-06:
In Home Assistant `sensor.desk_height` and `binary_sensor.desk_moving` never being updated after connection.

for moving I use `cover.open_cover`, `cover.close_cover`, `cover.stop_cover` and  `cover.set_cover_position` services

My config
```yaml
esphome:
  name: linak_desk
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "***"
  password: "***"
  fast_connect: true

idasen_desk_controller:
    mac_address: "00:00:00:00:00:00:00"

cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  - platform: idasen_desk_controller
    desk_height:
      # Height in cm
      name: "Desk Height"

binary_sensor:
  - platform: idasen_desk_controller
    name: "Desk Connection"
    type: CONNECTION
  - platform: idasen_desk_controller
    name: "Desk Moving"
    type: MOVING

captive_portal:

logger:

api:

ota:

```

there is logs
when use `cover.open_cover`
```
[13:57:31][D][cover:072]: 'Desk' - Setting
[13:57:31][D][cover:080]:   Position: 100%
[13:57:31][D][idasen_desk_controller:199]: Cover control - START - position 1.000000 - target 65 - current 18
[13:57:31][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 65
[13:57:31][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 65
[13:57:31][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 65
[13:57:32][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 65
[13:57:32][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 65
[13:57:32][D][cover:072]: 'Desk' - Setting
[13:57:32][D][cover:076]:   Command: STOP
[13:57:32][D][idasen_desk_controller:212]: Cover control - STOP
```
when use `cover.set_cover_position`
```
[14:00:49][D][idasen_desk_controller:199]: Cover control - START - position 0.200000 - target 13 - current 19
[14:00:49][D][idasen_desk_controller:232]: Update Desk - Move from 19 to 13
[14:00:49][D][idasen_desk_controller:232]: Update Desk - Move from 19 to 13
[14:00:49][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 18 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 17 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 17 to 13
[14:00:50][D][idasen_desk_controller:232]: Update Desk - Move from 16 to 13
[14:00:51][D][idasen_desk_controller:232]: Update Desk - Move from 15 to 13
[14:00:51][D][idasen_desk_controller:232]: Update Desk - Move from 15 to 13
[14:00:51][D][idasen_desk_controller:232]: Update Desk - Move from 14 to 13
[14:00:51][D][idasen_desk_controller:232]: Update Desk - Move from 14 to 13
[14:00:51][D][idasen_desk_controller:227]: Update Desk - target reached
```

j5lien, 2021-02-06:
Could you share you configuration logs, something like: 

```
[15:47:40][C][idasen_desk_controller:106]: Idasen Desk Controller:
[15:47:40][C][idasen_desk_controller:107]:   Mac address: 00:00:00:00:00
[15:47:40][C][idasen_desk_controller:108]:   Desk height 'Desk Height'
[15:47:40][C][idasen_desk_controller:108]:     Unit of Measurement: 'cm'
[15:47:40][C][idasen_desk_controller:108]:     Accuracy Decimals: 0
[15:47:40][C][idasen_desk_controller:108]:     Icon: 'mdi:arrow-up-down'
[15:47:40][C][idasen_desk_controller:109]:   Desk moving 'Desk Moving'
[15:47:40][C][idasen_desk_controller:109]:     Device Class: 'moving'
[15:47:40][C][idasen_desk_controller:110]:   Desk connection Desk Connection'
[15:47:40][C][idasen_desk_controller:110]:     Device Class: 'connectivity'
[15:47:40][C][idasen_desk_controller:111]:   Desk 'Desk'

IhorSyerkov, 2021-02-07:
```
[10:43:13][C][idasen_desk_controller:106]: Idasen Desk Controller:
[10:43:13][C][idasen_desk_controller:107]:   Mac address: 00:00:00:00:00:00
[10:43:13][C][idasen_desk_controller:108]:   Desk height 'Desk Height'
[10:43:13][C][idasen_desk_controller:108]:     Unit of Measurement: 'cm'
[10:43:13][C][idasen_desk_controller:108]:     Accuracy Decimals: 0
[10:43:13][C][idasen_desk_controller:108]:     Icon: 'mdi:arrow-up-down'
[10:43:13][C][idasen_desk_controller:109]:   Desk moving 'Desk Moving'
[10:43:13][C][idasen_desk_controller:109]:     Device Class: 'moving'
[10:43:13][C][idasen_desk_controller:110]:   Desk connection 'Desk Connection'
[10:43:13][C][idasen_desk_controller:110]:     Device Class: 'connectivity'
[10:43:13][C][idasen_desk_controller:111]:   Desk 'Desk'
```

j5lien, 2021-02-07:
So the `sensor.desk_height` is always `0` even after a restart ?

j5lien, 2021-02-07:
A notification callback is supposed to be called when the desk is moving but it didn't seem to work in your case. When the desk is moving, is there a led flashing continuously on your ESP32 ?

IhorSyerkov, 2021-02-07:
`sensor.desk_height` have some value. I guest it sets right after connect. When desk is moving led on esp not flashing, it allays on.

IhorSyerkov, 2021-02-07:
I added debug log in `deskHeightUpdateNotificationCallback` func and looks like it never called

IhorSyerkov, 2021-02-08:
Add second parameter to `registerForNotify` fixed issue for me. Now notifications works (height and moving)
```cpp
this->m_output_char_->registerForNotify(deskHeightUpdateNotificationCallback, true);
```
As I found, second parameter changes a value for descriptor
```cpp
uint8_t val[] = {0x01, 0x00};
if(!notifications) val[0] = 0x02;
BLERemoteDescriptor* desc = getDescriptor(BLEUUID((uint16_t)0x2902));
desc->writeValue(val, 2);
```
Not sure is only issue for my case. As I understand for you @j5lien everything works without this parameter?!

j5lien, 2021-02-13:
I thought this argument was set to true by default but it can be forced to this value.

j5lien, 2021-02-13:
I hope #5 will prevent this issue

closed 2021-02-13 by j5lien