j5lien/esphome-idasen-desk-controller #7: retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown

ACrazyConcept, 2021-02-14:
Hi

I am trying to use this with a [Linak GPD1M](https://www.linak.com/segments/deskline/start/dpg1m/). I figured the control would be similar enough to have it work. But I get the following error with very minor variations and it never gets past this point:

```
[19:37:32][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[19:37:37][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[19:37:37][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[19:37:37][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[19:37:37][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[19:37:37][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[19:37:37][E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown
[19:37:37]Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
[19:37:37]Core 0 register dump:
[19:37:37]PC      : 0x00000000  PS      : 0x00060730  A0      : 0x8016a0f6  A1      : 0x3ffeb370
[19:37:37]A2      : 0x3ffeb3b4  A3      : 0x3fff300c  A4      : 0x00060120  A5      : 0x00000000
[19:37:37]A6      : 0x00000001  A7      : 0x00000026  A8      : 0x80198ca7  A9      : 0x3ffeb320
[19:37:37]A10     : 0x3fff300c  A11     : 0x00000270  A12     : 0x00000000  A13     : 0x3fff3030
[19:37:37]A14     : 0x3fff2fbc  A15     : 0x00000009  SAR     : 0x00000010  EXCCAUSE: 0x00000014
[19:37:37]EXCVADDR: 0x00000000  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000
[19:37:37]
[19:37:37]Backtrace: 0x00000000:0x3ffeb370 0x4016a0f3:0x3ffeb390 0x40170273:0x3ffeb3b0 0x4016ed2a:0x3ffeb3d0 0x4018f445:0x3ffeb3f0 0x4018ca4e:0x3ffeb680 0x4018e1fd:0x3ffeb6a0 0x4018f457:0x3ffeb930 0x4018ca4e:0x3ffebbc0 0x4018e1fd:0x3ffebbe0 0x4018f457:0x3ffebe70 0x4018ca4e:0x3ffec100 0x4018e1fd:0x3ffec120 0x4018f457:0x3ffec3b0 0x4018ca4e:0x3ffec640 0x4018e1fd:0x3ffec660 0x4018f457:0x3ffec8f0 0x4018ca4e:0x3ffecb80 0x4018cb8d:0x3ffecba0 0x4019516a:0x3ffecbc0 0x40169fce:0x3ffecbe0 0x4008f1a5:0x3ffecc10
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4016a0f3: btu_free_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:504
WARNING Decoded 0x40170273: gatt_clcb_dealloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761
WARNING Decoded 0x4016ed2a: GATTC_Write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c:1040
WARNING Decoded 0x4018f445: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1163
WARNING Decoded 0x4018ca4e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4018e1fd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x4018f457: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4018ca4e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4018e1fd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x4018f457: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4018ca4e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4018e1fd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x4018f457: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4018ca4e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4018e1fd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x4018f457: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4018ca4e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4018cb8d: bta_gattc_hdl_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:397
WARNING Decoded 0x4019516a: bta_sys_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/sys/bta_sys_main.c:496
WARNING Decoded 0x40169fce: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:233
WARNING Decoded 0x4008f1a5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[19:37:38]
[19:37:38]Rebooting...
```

IhorSyerkov, 2021-02-15:
I would recommend you to play with your table using BT tools like https://github.com/jnross/Bluetility .
Check is services and characteristic are the same.
```cpp
static BLEUUID outputServiceUUID("99fa0020-338a-1024-8a49-009c0215f78a");
static BLEUUID outputCharacteristicUUID("99fa0021-338a-1024-8a49-009c0215f78a");
static BLEUUID inputServiceUUID("99fa0030-338a-1024-8a49-009c0215f78a");
static BLEUUID inputCharacteristicUUID("99fa0031-338a-1024-8a49-009c0215f78a");
static BLEUUID controlServiceUUID("99fa0001-338a-1024-8a49-009c0215f78a");
static BLEUUID controlCharacteristicUUID("99fa0002-338a-1024-8a49-009c0215f78a");
```

hsolberg, 2021-02-15:
Hi!

I'm having similar issues with the official Linak Bluetooth adapter (BLE2LIN). [Product Link](https://www.linak.no/produkter/tilbehoer/bluetooth-adapter-desk/#/description)

Here's the logs before it reboots.
````
[16:19:08][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[16:19:08][C][idasen_desk_controller:129]: Retrieve input remote characteristic.
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][C][idasen_desk_controller:134]: Retrieve output remote characteristic.
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:19:10][E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown
[16:19:10][C][idasen_desk_controller:137]: Register notification callback on output characteristic.
[16:19:10][C][idasen_desk_controller:143]: Retrieve control remote characteristic.
[16:19:10]***ERROR*** A stack overflow in task btuT has been detected.
[16:19:10]abort() was called at PC 0x400923c0 on core 0
[16:19:10]
[16:19:10]Backtrace: 0x40092178:0x3ffe8c00 0x400923a9:0x3ffe8c20 0x400923c0:0x3ffe8c40 0x4008f48c:0x3ffe8c60 0x400910f0:0x3ffe8c80 0x400910a6:0x00000000
[16:19:10]
[16:19:10]Rebooting...
````

After the reboot before running setup and crashing with stack overflow again (loops until it goes into safe-mode):
````
[16:19:10]ets Jun  8 2016 00:22:57
[16:19:10]
[16:19:10]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:19:10]configsip: 0, SPIWP:0xee
[16:19:10]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:19:10]mode:DIO, clock div:2
[16:19:10]load:0x3fff0018,len:4
[16:19:10]load:0x3fff001c,len:1044
[16:19:10]load:0x40078000,len:8896
[16:19:10]load:0x40080400,len:5828
[16:19:10]entry 0x400806ac
[16:19:10][I][logger:166]: Log initialized
[16:19:10][C][ota:366]: There have been 3 suspected unsuccessful boot attempts.
[16:19:10][I][app:029]: Running through setup()...
````

Also I noticed this warning during compiling if it's relevant.
````
/root/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteService.cpp: In member function 'void BLERemoteService::getCharacteristics(std::map<short unsigned int, BLERemoteCharacteristic*>*)':
/root/.platformio/packages/framework-arduinoespressif32/libraries/BLE/src/BLERemoteService.cpp:246:89: warning: parameter 'pCharacteristicMap' set but not used [-Wunused-but-set-parameter]
 void BLERemoteService::getCharacteristics(std::map<uint16_t, BLERemoteCharacteristic*>* pCharacteristicMap) {
                                                                                         ^
````

ACrazyConcept, 2021-02-15:
Hi @IhorSyerkov
Unfortunately I do not have a mac. But I tried Bluetooth LE Explorer from the Microsoft Store
It looks like all the same UUIDs are there. Clicking on a charactaristic I can get the descriptor name, handle and value.

[image]

and in text with some manual screen capture and copy paste...

```
Service Name: GenericAccess
Service UUID: 00001800-0000-1000-8000-00805f9b34fb
  Characteristic Name: DeviceName - User Description: - Handle: 2 - Value: Dencon Delta 
  Characteristic Name: Appearance - User Description: - Handle: 4 - Value: 00-00
  Characteristic Name: PeripheralPreferredConnectionParameters - User Description: - Handle: 6 - Value: 10-00-30-00-00-00-32-00

Service Name: GenericAttribute
Service UUID: 00001801 -0000-1000-8000-00805f9b34fb
  Characteristic Name: ServiceChanged - User Description: - Handle: 9 - Value: Read Not Permitted

Service Name: 99fa0001-338a-1024-8a49-009c0215f78a
Service UUID: 99fa0001-338a-1024-8a49-009c0215f78a
  Characteristic Name: 99fa0002-338a-1024-8a49-009c0215f78a	- User Description:	-	Handle:	13	-	Value:	Read Not Permitted
  Characteristic Name: 99fa0003-338a-1024-8a49-009c0215f78a	- User Description:	-	Handle:	15	-	Value:	NULL

Service Name: 99fa0010-338a-1024-8a49-009c0215f78a
Service UUID: 99fa0010-338a-1024-8a49-009c0215f78a
  Characteristic Name: 99fa0011-338a-1024-8a49-009c0215f78a - User Description: - Handle: 19 - Value: DB

Service Name: Deviceinformation
Service UUID: 0000180a-0000-1000-8000-00805f9b34fb
  Characteristic Name: ManufacturerNameString - User Description: - Handle: 23 - Value: LINAKA/S 
  Characteristic Name: ModelNumberString - User Description: - Handle: 25 - Value: DPG

Service Name: 99fa0020-338a-1024-8a49-009c0215f78a
Service UUID: 99fa0020-338a-1024-8a49-009c0215f78a
  Characteristic Name: 99fa0021-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	28	- Value:	E0-04-00-00
  Characteristic Name: 99fa0022-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	31	- Value:	00-00-01-00
  Characteristic Name: 99fa0023-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	34	- Value:	00-00-01-00
  Characteristic Name: 99fa0024-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	37	- Value:	00-00-01-00
  Characteristic Name: 99fa0025-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	40	- Value:	00-00-01-00
  Characteristic Name: 99fa0026-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	43	- Value:	00-00-01-00
  Characteristic Name: 99fa0027-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	46	- Value:	00-00-01-00
  Characteristic Name: 99fa0028-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	49	- Value:	00-00-01-00
  Characteristic Name: 99fa0029-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	52	- Value:	01
  Characteristic Name: 99fa002a-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	54	- Value:	00

Service Name: 99fa0030-338a-1024-8a49-009c0215f78a
Service UUID: 99fa0030-338a-1024-8a49-009c0215f78a
Characteristic Name: 99fa0031-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	57	- Value:	Read Not Permitted
Characteristic Name: 99fa0032-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	59	- Value:	Read Not Permitted
Characteristic Name: 99fa0033-338a-1024-8a49-009c0215f78a - User Description:	- Handle:	61	- Value:	Read Not Permitted
Characteristic Name: 99fa0034-338a-1024-8a49-009c0215f78a - User Description: - Handle: 63  - Value:  Read Not Permitted
```

ACrazyConcept, 2021-02-15:
> I'm having similar issues with the official Linak Bluetooth adapter (BLE2LIN). [Product Link](https://www.linak.no/produkter/tilbehoer/bluetooth-adapter-desk/#/description)
> 
@hsolberg
I have actually considered getting that one myself too.

j5lien, 2021-02-20:
It seems to crash when retrieving the control characteristic while it is available in your logs

```
Service Name: 99fa0001-338a-1024-8a49-009c0215f78a
Service UUID: 99fa0001-338a-1024-8a49-009c0215f78a
  Characteristic Name: 99fa0002-338a-1024-8a49-009c0215f78a	- User Description:	-	Handle:	13	-	Value:	Read Not Permitted
  Characteristic Name: 99fa0003-338a-1024-8a49-009c0215f78a	- User Description:	-	Handle:	15	-	Value:	NULL
```

I don't really know how to help you on this issue

hsolberg, 2021-02-24:
@j5lien [This NodeJS Module](https://github.com/alex20465/deskbluez) worked for me, but that requires a Raspberry Pi or similar to run it, I would much prefer to use an ESP32 instead. Maybe there's some logic/handling there that might be of use? Here's the same info as @ACrazyConcept just for the official Linak Bluetooth adapter (BLE2LIN) if that helps at all (i only removed the mac-address). If this doesn't help at all, is it possible to add additional logging to uncover more details that might help?

````
Number of Services: 9
BT 4.2 Secure Connection: False

Service Name: GenericAccess
Service UID: 00001800-0000-1000-8000-00805f9b34fb

Characteristic Name: DeviceName - User Description: - Handle: 2 - Value: BLE2LIN 6671
Characteristic Name: Appearance - User Description: - Handle: 4 - Value: 00-00

Characteristic Name: PeripheralPreferredConnectionParameters - User Description: - Handle: 6 - Value: 10-00-30-00-00-00-32-00
Service Name: GenericAttribute
Service UID: 00001801-0000-1000-8000-00805f9b34fb
Characteristic Name: ServiceChanged - User Description: - Handle: 9 - Value: Unreachable
Service Name: 99fa0001-338a-1024-849-009c0215f78a
Service UID: 99fa0001-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0002-338a-1024-8a49-009c0215f78a - User Description: - Handle: 13 - Value: Unreachable

Characteristic Name: 99fa0003-338a-1024-8249-009c0215f78a - User Description: - Handle: 15 - Value: Unreachable

Service Name: 99fa0010-338a-1024-849-009c0215f78a
Service UID: 99fa0010-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0011-338a-1024-8a49-009c0215f78a - User Description: - Handle: 19 - Value: DF
Service Name: Devicelnformation

Service UID: 0000180a-0000-1000-8000-00805f9b34fb

Characteristic Name: ManufacturerNameString - User Description: - Handle: 23 - Value: LINAK A/S

Characteristic Name: ModelNumberString - User Description: - Handle: 25 - Value: BLE2LINOO1

Service Name: 99fa0020-338a-1024-849-009c0215f78a
Service UID: 99fa0020-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0021-338a-1024-8249-009c0215f78a - User Description: - Handle: 28 - Value: 80-02-00-00

Characteristic Name: 99fa0022-338a-1024-8249-009c0215f78a - User Description: - Handle: 31 - Value: 00-00-01-00

Characteristic Name: 99fa0023-338a-1024-8249-009c0215f78a - User Description: - Handle: 34 - Value: 00-00-01-00

Characteristic Name: 99fa0024-338a-1024-8249-009c0215f78a - User Description: - Handle: 37 - Value: 00-00-01-00

Characteristic Name: 99fa0025-338a-1024-8249-009c0215f78a - User Description: - Handle: 40 - Value: 00-00-01-00

Characteristic Name: 99fa0026-338a-1024-8249-009c0215f78a - User Description: - Handle: 43 - Value: 00-00-01-00

Characteristic Name: 99fa0027-338a-1024-8249-009c0215f78a - User Description: - Handle: 46 - Value: 00-00-01-00

Characteristic Name: 99fa0028-338a-1024-8249-009c0215f78a - User Description: - Handle: 49 - Value: 00-00-01-00

Characteristic Name: 99fa0029-338a-1024-8249-009c0215f78a - User Description: - Handle: 52 - Value: 01

Characteristic Name: 99fa002a-338a-1024-8a49-009c0215f78a - User Description: - Handle: 54 - Value: 00

Service Name: 99fa0030-338a-1024-849-009c0215f78a
Service UID: 99fa0030-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0031-338a-1024-8249-009c0215f78a - User Description: - Handle: 57 - Value: DE-12

Characteristic Name: 99fa0032-338a-1024-8249-009c0215f78a - User Description: - Handle: 59 - Value: Unreachable

Characteristic Name: 99fa0033-338a-1024-8249-009c0215f78a - User Description: - Handle: 61 - Value: Unreachable

Characteristic Name: 99fa0034-338a-1024-8249-009c0215f78a - User Description: - Handle: 63 - Value: Unreachable

Service Name: 99fa0040-338a-1024-849-009c0215f78a
Service UID: 99fa0040-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0041-338a-1024-8249-009c0215f78a - User Description: - Handle: 66 - Value: Unreachable

Characteristic Name: 99fa0043-338a-1024-8249-009c0215f78a - User Description: - Handle: 69 - Value: 00

Service Name: 99fa0050-338a-1024-8249-009c0215f78a
Service UID: 99fa0050-338a-1024-8249-009c0215f78a

Characteristic Name: 99fa0051-338a-1024-8249-009c0215f78a - User Description: - Handle: 72 - Value: Unreachable
````

j5lien, 2021-06-28:
On the latest release, I added an option to use different bluetooth control commands.
https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/README.md#not-moving-using-cover-component

Does it fix your issues ?

lars-1, 2021-06-28:
It does not crash anymore, but it won't connect to my desk:

```
[C][idasen_desk_controller:122]: Connecting client to device <removed MAC>
[C][idasen_desk_controller:126]: Fail to connect to client

-- pressed the bluetooth button on the desk here --

[C][idasen_desk_controller:122]: Connecting client to device <removed MAC>
[D][binary_sensor:036]: 'ESP32 Desk - Connection': Sending state ON
[C][idasen_desk_controller:132]: Retrieve input remote characteristic.
[C][idasen_desk_controller:132]: Retrieve input remote characteristic.
[C][idasen_desk_controller:137]: Retrieve output remote characteristic.
[C][idasen_desk_controller:148]: Retrieve control remote characteristic.
[C][idasen_desk_controller:153]: Success connecting client to device
[C][idasen_desk_controller:126]: Fail to connect to client
[C][idasen_desk_controller:122]: Connecting client to device <removed MAC>
```

I tried this config:
```
esp32_ble_tracker:

ble_client:
  - mac_address: "<removed MAC>"
    id: linak_desk
    
    
idasen_desk_controller:
    ble_client_id: linak_desk
    only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "${plug_name} - Desk"
```
and this config: https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/README.md#extra-desk-informations

j5lien, 2021-07-20:
Logs are coming from from the 1.x release of this component. Try to use the last one https://github.com/j5lien/esphome-idasen-desk-controller/releases/tag/v2.1.0

lars-1, 2021-07-23:
> Logs are coming from from the 1.x release of this component. Try to use the last one https://github.com/j5lien/esphome-idasen-desk-controller/releases/tag/v2.1.0

My configuration contains
```
external_components:
- source: github://j5lien/esphome-idasen-desk-controller@v2.1.0
```
so this should be the latest version 🤔

j5lien, 2021-07-25:
Yes it's the latest. Could tried to set up only the height sensor like I explain here https://github.com/j5lien/esphome-idasen-desk-controller/issues/25#issuecomment-883697499 without using the custom component ?

Do you still get connection issue ?

lars-1, 2021-07-25:
> Yes it's the latest. Could tried to set up only the height sensor like I explain here [#25 (comment)](https://github.com/j5lien/esphome-idasen-desk-controller/issues/25#issuecomment-883697499) without using the custom component ?
> 
> Do you still get connection issue ?

Yes, see attached logs. Looks like it connects but then disconnects immediately. Web UI does not show a height value.

```
[15:23:31][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[15:23:39][W][ble_sensor:039]: [Desk Height] Disconnected!
[15:23:39][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[15:23:39][W][ble_client:115]: connect to DE:SK:MA:CX:XX:XX failed, status=133
[15:23:39][D][esp32_ble_tracker:180]: Starting scan...
[15:23:39][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-62
[15:23:39][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[15:23:40][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-77
[15:23:40][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[15:23:40][D][esp32_ble_tracker:645]:   TX Power: 2
(removed some devices here...)
[15:23:40][D][ble_client:045]: Found device at MAC address [DE:SK:MA:CX:XX:XX]
[15:23:40][I][ble_client:083]: Attempting BLE connection to DE:SK:MA:CX:XX:XX
[15:23:40][I][ble_sensor:033]: [Desk Height] Connected successfully!
[15:23:42][I][ble_client:159]: Service UUID: 0x1800
[15:23:42][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[15:23:42][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[15:23:42][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[15:23:42][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[15:23:42][I][ble_client:159]: Service UUID: 0x1801
[15:23:42][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[15:23:42][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[15:23:42][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[15:23:42][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[15:23:42][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0xe, properties 0xc
[15:23:42][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0x12
[15:23:42][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[15:23:42][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[15:23:42][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x14, properties 0x1e
[15:23:42][I][ble_client:159]: Service UUID: 0x180A
[15:23:42][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[15:23:42][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[15:23:42][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[15:23:42][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[15:23:42][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[15:23:42][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0022338A10-248A-4900-9C02-15F78A, handle 0x20, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0023338A10-248A-4900-9C02-15F78A, handle 0x23, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0024338A10-248A-4900-9C02-15F78A, handle 0x26, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0025338A10-248A-4900-9C02-15F78A, handle 0x29, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0026338A10-248A-4900-9C02-15F78A, handle 0x2c, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0027338A10-248A-4900-9C02-15F78A, handle 0x2f, properties 0x12
[15:23:42][I][ble_client:339]:  characteristic 99FA0028338A10-248A-4900-9C02-15F78A, handle 0x32, properties 0x12
[15:23:43][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x35, properties 0x2
[15:23:43][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x37, properties 0x2
[15:23:43][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[15:23:43][I][ble_client:160]:   start_handle: 0x38  end_handle: 0xffff
[15:23:43][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x3a, properties 0xc
[15:23:43][I][ble_client:339]:  characteristic 99FA0032338A10-248A-4900-9C02-15F78A, handle 0x3c, properties 0xc
[15:23:43][I][ble_client:339]:  characteristic 99FA0033338A10-248A-4900-9C02-15F78A, handle 0x3e, properties 0xc
[15:23:43][I][ble_client:339]:  characteristic 99FA0034338A10-248A-4900-9C02-15F78A, handle 0x40, properties 0xc
[15:23:43][D][esp32_ble_tracker:180]: Starting scan...
[15:23:43][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-76
[15:23:43][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[15:23:43][D][esp32_ble_tracker:645]:   TX Power: 2
[15:23:43][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-77
[15:23:43][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[15:23:43][D][esp32_ble_tracker:645]:   TX Power: 2
[15:23:43][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-57
[15:23:43][D][esp32_ble_tracker:641]:   Address Type: RANDOM
(removed some devices here...)
[15:24:01][W][ble_sensor:082]: Error reading char at handle 29, status=5
[15:24:07][D][esp32_ble_tracker:620]: Found device XX:XX:XX:XX:XX:XX RSSI=-73
[... cut ...]
[15:24:18][D][ble_client:045]: Found device at MAC address [DE:SK:MA:CX:XX:XX]
[15:24:18][I][ble_client:083]: Attempting BLE connection to DE:SK:MA:CX:XX:XX
[15:24:31][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[15:24:48][W][ble_sensor:039]: [Desk Height] Disconnected!
[15:24:48][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[15:24:48][W][ble_client:115]: connect to DE:SK:MA:CX:XX:XX failed, status=133
[15:24:48][D][esp32_ble_tracker:180]: Starting scan...
```
removed MAC addresses for privacy