j5lien/esphome-idasen-desk-controller #9: No height / moving sensor updates

noxhirsch, 2021-02-17:
The height and moving sensors don't update their values in Home Assistant. The height sensor always shows 14cm and the moving sensor "not moving" - everything else seems to work perfectly fine.

My config is exactly the example code from README.md (exept the names) and my log doesn't show anything weird:

<details>
  <summary>ESPHome Log</summary>
  
  ```
[21:29:07][C][idasen_desk_controller:096]: Idasen Desk Controller:
[21:29:07][C][idasen_desk_controller:097]:   Mac address: XX:XX:XX:XX:XX:XX
[21:29:07][C][idasen_desk_controller:098]:   Desk height 'AZ Desk Workstation Height'
[21:29:07][C][idasen_desk_controller:098]:     Unit of Measurement: 'cm'
[21:29:07][C][idasen_desk_controller:098]:     Accuracy Decimals: 0
[21:29:07][C][idasen_desk_controller:098]:     Icon: 'mdi:arrow-up-down'
[21:29:07][C][idasen_desk_controller:099]:   Desk moving 'AZ Desk Workstation Moving'
[21:29:07][C][idasen_desk_controller:099]:     Device Class: 'moving'
[21:29:07][C][idasen_desk_controller:100]:   Desk connection 'AZ Desk Workstation Connection'
[21:29:07][C][idasen_desk_controller:100]:     Device Class: 'connectivity'
[21:29:07][C][idasen_desk_controller:101]:   Desk 'AZ Desk Workstation'
[21:29:07][C][idasen_desk_controller:109]: Start scanning devices...
[21:29:07][C][idasen_desk_controller:052]: BLE Device found: Name: Desk Caro, Address: XXXXXX, txPower: -8
[21:29:07][C][idasen_desk_controller:052]: BLE Device found: Name: , Address: 192.168.XX.XX, manufacturer data: 4c001005471c7e6b36, txPower: 12
[21:29:07][C][idasen_desk_controller:052]: BLE Device found: Name: Desk Nico, Address: 192.168.XX.XX, txPower: -8

[21:40:46][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:46][D][cover:080]:   Position: 0%
[21:40:46][D][idasen_desk_controller:180]: Cover control - START - position 0.000000 - target 0 - current 53
[21:40:46][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:46][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:46][D][cover:076]:   Command: STOP
[21:40:46][D][idasen_desk_controller:191]: Cover control - STOP
[21:40:49][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:49][D][cover:080]:   Position: 0%
[21:40:49][D][idasen_desk_controller:180]: Cover control - START - position 0.000000 - target 0 - current 53
[21:40:49][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:49][D][cover:076]:   Command: STOP
[21:40:49][D][idasen_desk_controller:191]: Cover control - STOP
[21:40:49][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:49][D][cover:080]:   Position: 0%
[21:40:49][D][idasen_desk_controller:180]: Cover control - START - position 0.000000 - target 0 - current 53
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 53 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 52 to 0
[21:40:50][D][idasen_desk_controller:209]: Update Desk - Move from 52 to 0
[21:40:51][D][idasen_desk_controller:209]: Update Desk - Move from 51 to 0
[21:40:51][D][idasen_desk_controller:209]: Update Desk - Move from 51 to 0
[21:40:51][D][idasen_desk_controller:209]: Update Desk - Move from 50 to 0
[21:40:51][D][idasen_desk_controller:209]: Update Desk - Move from 50 to 0
[21:40:51][D][idasen_desk_controller:209]: Update Desk - Move from 49 to 0
[21:40:51][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:51][D][cover:076]:   Command: STOP
[21:40:51][D][idasen_desk_controller:191]: Cover control - STOP
[21:40:52][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:40:52][D][cover:080]:   Position: 100%
[21:40:52][D][idasen_desk_controller:180]: Cover control - START - position 1.000000 - target 65 - current 48
[21:40:52][D][idasen_desk_controller:209]: Update Desk - Move from 48 to 65
[21:40:52][D][idasen_desk_controller:209]: Update Desk - Move from 48 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 48 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 48 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 49 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 49 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 49 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 49 to 65
[21:40:53][D][idasen_desk_controller:209]: Update Desk - Move from 50 to 65
[21:40:54][D][idasen_desk_controller:209]: Update Desk - Move from 50 to 65
[21:40:54][D][idasen_desk_controller:209]: Update Desk - Move from 51 to 65
[21:40:54][D][idasen_desk_controller:209]: Update Desk - Move from 51 to 65
[... 113 lines cut ...]
[21:41:53][D][idasen_desk_controller:209]: Update Desk - Move from 23 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 22 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 22 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 21 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 21 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 20 to 14
[21:41:54][D][idasen_desk_controller:209]: Update Desk - Move from 19 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 19 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 18 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 18 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 17 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 17 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 16 to 14
[21:41:55][D][idasen_desk_controller:209]: Update Desk - Move from 15 to 14
[21:41:56][D][idasen_desk_controller:209]: Update Desk - Move from 15 to 14
[21:41:56][D][idasen_desk_controller:204]: Update Desk - target reached
[21:42:15][D][api:067]: Disconnecting Home Assistant 2021.2.3 ()
[21:42:44][D][cover:072]: 'AZ Desk Workstation' - Setting
[21:42:44][D][cover:080]:   Position: 22%
[21:42:44][D][idasen_desk_controller:180]: Cover control - START - position 0.220000 - target 14 - current 14
  ```

</details>

Do you have an idea, what else I could look for?

spyfly, 2021-02-18:
I am experiencing the same issue.

mountainsandcode, 2021-02-20:
Can confirm this issue, too.

mountainsandcode, 2021-02-20:
Appears to be a duplicate of #2, but I can confirm this is an issue with the latest master

j5lien, 2021-02-20:
Do you find the log `Register notification callback`  when connecting to the desk ?
```
[14:38:13][C][idasen_desk_controller:137]: Register notification callback on output characteristic.
[14:38:13][C][idasen_desk_controller:143]: Retrieve control remote characteristic.
[14:38:13][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[14:38:13][C][idasen_desk_controller:148]: Success connecting client to device
```

I'm not sure why but it doesn't seem to work at first.

noxhirsch, 2021-02-20:
@j5lien Yes, here's the log:
```
[16:05:07][C][idasen_desk_controller:109]: Start scanning devices...
[16:05:07][C][idasen_desk_controller:052]: BLE Device found: Name: , Address: XXXXXXXXXXXXXXXXXXXXXXXXXXXXX, serviceUUID: XXXXXXXXXXXXXXXXXXXXXXXXXXXXX
[16:05:07][C][idasen_desk_controller:052]: BLE Device found: Name: Desk Nico, Address: XXXXXXXXXXXXXXXXXXXXXXX, txPower: -8
[16:05:07][C][idasen_desk_controller:119]: Connecting client to device XXXXXXXXXXXXXXXXXXXXXXX, 
[16:05:07][D][binary_sensor:036]: 'AZ Desk Workstation Connection': Sending state ON
[16:05:07][C][idasen_desk_controller:129]: Retrieve input remote characteristic.
[16:05:08][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:05:08][C][idasen_desk_controller:134]: Retrieve output remote characteristic.
[16:05:08][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:05:08][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:05:08][C][idasen_desk_controller:137]: Register notification callback on output characteristic.
[16:05:08][C][idasen_desk_controller:143]: Retrieve control remote characteristic.
[16:05:08][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[16:05:08][C][idasen_desk_controller:148]: Success connecting client to device
[16:05:08][D][api.connection:604]: Client 'Home Assistant 2021.2.3 (XXXXXXXXX)' connected successfully!
[16:05:13][D][binary_sensor:036]: 'AZ Desk Workstation Moving': Sending state OFF
[16:05:13][D][sensor:092]: 'AZ Desk Workstation Height': Sending state 143.00000 mm with 0 decimals of accuracy
[16:05:13][D][cover:152]: 'AZ Desk Workstation' - Publishing:
[16:05:13][D][cover:155]:   Position: 22%
[16:05:13][D][cover:168]:   Current Operation: IDLE
```

j5lien, 2021-02-20:
So the callback is well registered but isn't called when the desk is moving 🤔

mountainsandcode, 2021-02-22:
When trying to setup a script myself I had issues getting callbacks to work - I think they may work differently than for some other LINAK desks but I’m not sure.

As a temporary workaround, can you manually update the position after it has completed moving? That would solve a big part of the problem for me

mountainsandcode, 2021-04-13:
@noxhirsch @spyfly You may want to try out the code from #12 to confirm whether this also fixes the issues for you - I'm having no issues using that code with bluetooth_callback set to False.

noxhirsch, 2021-04-13:
@mountainsandcode For me it's working with `bluetooth_callback: false`, too 👍

closed 2021-04-14 by j5lien