j5lien/esphome-idasen-desk-controller #10: Incompatibility with esp32_ble_tracker

spyfly, 2021-02-18:
It seems like there is an incompatibility with esp32_ble_tracker, which causes theESP32 to reboot every couple minutes.

Here's the log output:
```
[23:44:39][W][esp32_ble_tracker:164]: ESP-IDF BLE scan never terminated, rebooting to restore BLE stack...
[23:44:39][I][app:132]: Forcing a reboot...
```

j5lien, 2021-02-20:
It doesn't use the same bluetooth libraries so it won't be compatible I think

spyfly, 2021-02-20:
> It doesn't use the same bluetooth libraries so it won't be compatible I think

Yeah, figured that out yesterday when looking into the codebase. I've deployed a dedicated ESP32 for the desk control to fix this issue for now.

However even if you use the same library, having a constantly active connection to the desk may also be the cause of this issue.

j5lien, 2021-02-20:
Possible, I keep an active connection to be notified when moving the desk from the controller.

j5lien, 2021-06-19:
It should be compatible with the new version

spyfly, 2021-06-20:
> It should be compatible with the new version

Amazing, now I don't have to run a second ESP32, just for my desk anymore :+1:

Thanks a lot for resolving this issue :heart:

closed 2021-06-20 by spyfly