j5lien/esphome-idasen-desk-controller #11: TypeError in sensor_schema reported as missing 1 required positional argument: 'device_class_'

mkotek, 2021-03-31:
When trying to define the ESP32 code, I am getting sensor definition issue. When running validation on the code, the issue is reported by problem with sensor.py file:
`  File "/config/esphome/custom_components/idasen_desk_controller/sensor.py", line 22, in <module>
    sensor.sensor_schema(UNIT_HEIGHT, ICON_HEIGHT, 0),
TypeError: sensor_schema() missing 1 required positional argument: 'device_class_'
Failed config`

Is it possible, something has been changed in the definition of HA sensor itself, that causes the issue? I am on HA core-2021.3.4 and ESPHOME 1.16.2.

mkotek, 2021-03-31:
For the sake of others, it seems, the new definition needs to be:
'    cv.Optional(CONF_HEIGHT):
        sensor.sensor_schema(UNIT_HEIGHT, ICON_HEIGHT, 0, ''),'

closed 2021-03-31 by mkotek

j5lien, 2021-04-10:
I think it's gonna be an issue with the next release of ESPHome 1.17

noxhirsch, 2021-05-05:
I can confirm that it is an issue with version 1.17, which got released yesterday.
@mkotek's solution solved the problem.

j5lien, 2021-05-06:
It has been fixed in 1.17.1 as the argument is now optional