j5lien/esphome-idasen-desk-controller #13: Add offset to desk height sensor

mountainsandcode, 2021-04-13:
Thanks again for your work on this component, @j5lien.

One idea specifically for the sensor.desk_height - I'm currently using a template sensor to add an offset (61cm) to that value to get the actual height of the table and use this throughout HA. As a potential future feature, it would be cool if the sensor.desk_height had a config value for an offset, so that the value of the esphome sensor has the right value right away.

Not a priority, just an idea :)

j5lien, 2021-04-14:
Hey @mountainsandcode, I think you can do this using sensor offset filter https://esphome.io/components/sensor/index.html#offset-multiply

```yaml
sensor:
  - platform: idasen_desk_controller
    desk_height:
      name: "Desk Height"
      filters:
        # It should add 50 to every value that are published
        - offset: 50.0
```

mountainsandcode, 2021-04-16:
@j5lien - Awesome, didn't know this was a feature in ESPHome!

closed 2021-04-16 by mountainsandcode