j5lien/esphome-idasen-desk-controller #17: Endless loop when calling cover.control in ESPHome or moving the desk manually

noxhirsch, 2021-04-18:
To move the desk with millimeter precision, I created a service, since moving the cover with float values doesn't seem to work with Home Assistant directly.
```
api:
  password: XXX
  services:
    - service: move_to_position
      variables:
        position: float
      then:
        - cover.control:
            id: idasen_desk
            position: !lambda 'return position/100;'
```
When moving the desk only a very small amount it moves to the target position and then keeps trying to move the desk in an endless loop. The same happens when the desk is already at the target position and also if the desk is stopped manually with the old "dumb" control switch while moving.

I think the easiest solution here would be to break the loop if the speed is 0 twice in a row.

Log:
```
[10:28:02][D][cover:072]: 'AZ Desk Nico' - Setting
[10:28:02][D][cover:080]:   Position: 22%
[10:28:02][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.3 - speed 0.0
[10:28:02][D][idasen_desk_controller:289]: Cover control - START - position 0.2 - target 14.6 - current 14.3
[10:28:02][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.3 - speed 0.0
[10:28:02][D][idasen_desk_controller:274]: Update Desk - Move from 14.3 to 14.6
[10:28:02][D][cover:152]: 'AZ Desk Nico' - Publishing:
[10:28:02][D][cover:155]:   Position: 22%
[10:28:02][D][cover:168]:   Current Operation: OPENING
[10:28:03][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.3 - speed 0.0
[10:28:03][D][idasen_desk_controller:274]: Update Desk - Move from 14.3 to 14.6
[10:28:03][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.3 - speed 12.0
[10:28:03][D][binary_sensor:036]: 'AZ Desk Nico Moving': Sending state ON
[10:28:03][D][idasen_desk_controller:274]: Update Desk - Move from 14.3 to 14.6
[10:28:03][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:03][D][sensor:092]: 'AZ Desk Nico Height': Sending state 76.60000 cm with 1 decimals of accuracy
[10:28:03][D][binary_sensor:036]: 'AZ Desk Nico Moving': Sending state OFF
[10:28:03][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
[10:28:03][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:03][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
[10:28:03][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:03][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
[10:28:04][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:04][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
[10:28:04][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:04][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
[10:28:04][D][idasen_desk_controller:199]: Desk bluetooth data: height 14.6 - speed 0.0
[10:28:04][D][idasen_desk_controller:274]: Update Desk - Move from 14.6 to 14.6
........
```

j5lien, 2021-06-19:
Is the new version based on ESPHome BLE Client fixes your issue ?

noxhirsch, 2021-06-19:
In the new version, the height sensor never updates and because of that, the script is stuck in an endless loop, too (but the desk stops at the right position if I set a target).

This is the log while and after moving to position:

```
[20:47:58][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:58][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:58][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:59][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:59][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:59][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:59][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:47:59][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:48:00][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
[20:48:00][D][idasen_desk_controller:194]: Update Desk - Move from 22 to 82
```

When adding an `update_interval`, the loop stops, but I don't know if it's the best solution to spam the bluetooth connection with update requests 24/7.

j5lien, 2021-06-19:
Do you see this error in your logs `esp_ble_gattc_register_for_notify failed` ?

noxhirsch, 2021-06-19:
No, I can't find this  error. I can only see multiple `Unhandled type: advType: 0x12`

Here's the complete log on verbose level:
<details>
  <summary>Complete Log</summary>
  
  ```
[22:20:52][I][app:105]: ESPHome version 1.19.1 compiled on Jun 19 2021, 22:19:45
[22:20:52][C][template.binary_sensor:018]: Template Binary Sensor 'AZ Desk Nico Connection'
[22:20:52][C][template.binary_sensor:018]: Template Binary Sensor 'AZ Desk Nico Moving'
[22:20:52][C][logger:189]: Logger:
[22:20:52][C][logger:190]:   Level: VERBOSE
[22:20:52][C][logger:191]:   Log Baud Rate: 115200
[22:20:52][C][logger:192]:   Hardware UART: UART0
[22:20:52][C][esp32_ble_tracker:606]: BLE Tracker:
[22:20:52][C][esp32_ble_tracker:607]:   Scan Duration: 300 s
[22:20:52][C][esp32_ble_tracker:608]:   Scan Interval: 320.0 ms
[22:20:52][C][esp32_ble_tracker:609]:   Scan Window: 30.0 ms
[22:20:52][C][esp32_ble_tracker:610]:   Scan Type: ACTIVE
[22:20:52][C][ble_client:033]: BLE Client:
[22:20:52][C][ble_client:034]:   Address: XX:XX:XX:XX:XX:XX
[22:20:52][C][ble_sensor:019]: BLE Sensor 'AZ Desk Nico Height'
[22:20:52][C][ble_sensor:019]:   State Class: ''
[22:20:52][C][ble_sensor:019]:   Unit of Measurement: 'cm'
[22:20:52][C][ble_sensor:019]:   Accuracy Decimals: 1
[22:20:52][C][ble_sensor:019]:   Icon: 'mdi:arrow-up-down'
[22:20:52][C][ble_sensor:020]:   MAC address        : XX:XX:XX:XX:XX:XX
[22:20:52][C][ble_sensor:021]:   Service UUID       : 99FA0020338A10-248A-4900-9C02-15F78A
[22:20:52][C][ble_sensor:022]:   Characteristic UUID: 99FA0021338A10-248A-4900-9C02-15F78A
[22:20:52][C][ble_sensor:023]:   Descriptor UUID    : 00000000000000-0000-0000-0000-000000
[22:20:52][C][ble_sensor:024]:   Notifications      : YES
[22:20:52][C][ble_sensor:025]:   Update Interval: 4294967.5s
[22:20:52][C][ble_sensor:019]: BLE Sensor 'AZ Desk Nico Speed'
[22:20:52][C][ble_sensor:019]:   State Class: ''
[22:20:52][C][ble_sensor:019]:   Unit of Measurement: 'cm/min'
[22:20:52][C][ble_sensor:019]:   Accuracy Decimals: 0
[22:20:52][C][ble_sensor:019]:   Icon: 'mdi:speedometer'
[22:20:52][C][ble_sensor:020]:   MAC address        : XX:XX:XX:XX:XX:XX
[22:20:52][C][ble_sensor:021]:   Service UUID       : 99FA0020338A10-248A-4900-9C02-15F78A
[22:20:52][C][ble_sensor:022]:   Characteristic UUID: 99FA0021338A10-248A-4900-9C02-15F78A
[22:20:52][C][ble_sensor:023]:   Descriptor UUID    : 00000000000000-0000-0000-0000-000000
[22:20:52][C][ble_sensor:024]:   Notifications      : YES
[22:20:52][C][ble_sensor:025]:   Update Interval: 4294967.5s
[22:20:52][C][ethernet:068]: Ethernet:
[22:20:52][C][ethernet:222]:   IP Address: 192.168.XXX.XXX
[22:20:52][C][ethernet:223]:   Hostname: 'az_desk_nico'
[22:20:52][C][ethernet:224]:   Subnet: 255.255.255.0
[22:20:52][C][ethernet:225]:   Gateway: 192.168.55.1
[22:20:52][C][ethernet:236]:   DNS1: 192.168.55.1
[22:20:52][C][ethernet:237]:   DNS2: 192.168.50.1
[22:20:52][C][ethernet:240]:   MAC Address: XX:XX:XX:XX:XX:XX
[22:20:52][C][ethernet:241]:   Is Full Duplex: YES
[22:20:52][C][ethernet:242]:   Link Up: YES
[22:20:52][C][ethernet:243]:   Link Speed: 100
[22:20:52][C][ethernet:070]:   Power Pin: GPIO16 (Mode: OUTPUT)
[22:20:52][C][ethernet:071]:   MDC Pin: 23
[22:20:52][C][ethernet:072]:   MDIO Pin: 18
[22:20:52][C][ethernet:073]:   Type: LAN8720
[22:20:53][C][ota:029]: Over-The-Air Updates:
[22:20:53][C][ota:030]:   Address: az_desk_nico.local:3232
[22:20:53][C][ota:032]:   Using Password.
[22:20:53][C][api:095]: API Server:
[22:20:53][C][api:096]:   Address: az_desk_nico.local:6053
[22:20:53][C][idasen_desk_controller:024]: Idasen Desk Controller:
[22:20:53][C][idasen_desk_controller:025]:   MAC address        : XX:XX:XX:XX:XX:XX
[22:20:53][C][idasen_desk_controller:026]:   Desk 'AZ Desk Nico'
[22:20:53][V][esp32_ble_tracker:591]: Unhandled type: advType: 0x12
[22:20:54][V][esp32_ble_tracker:591]: Unhandled type: advType: 0x12
[22:20:54][V][esp32_ble_tracker:591]: Unhandled type: advType: 0x12
[22:20:55][V][esp32_ble_tracker:591]: Unhandled type: advType: 0x12
[... 77 lines cut ...]
[22:21:19][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[22:21:19][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:19][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:19][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:19][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:20][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:20][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:20][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:20][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:20][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:21][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:21][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:21][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:21][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:21][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:22][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:22][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:22][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:22][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:22][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[22:21:23][D][idasen_desk_controller:194]: Update Desk - Move from 82 to 22
[...]
  ```
  
</details>

spyfly, 2021-06-21:
Still seems to be an issue in v2.0.1
[image]

j5lien, 2021-06-28:
@noxhirsch is the v2.0.1 fixes some of your problems ?

@spyfly could you share logs after your device just booted ?

spyfly, 2021-06-29:
> @spyfly could you share logs after your device just booted ?
It seems like the issue has been resolved. Desk Height now properly updates both when changing it manually or via HA.

j5lien, 2021-07-20:
@noxhirsch Have you try the latest release with the option `only_up_down_command` set to `true` ?
https://github.com/j5lien/esphome-idasen-desk-controller#not-moving-using-cover-component

noxhirsch, 2021-07-21:
@j5lien I just tried that out. Unfortunately, it only made it worse. The height is not updated here either and the desk moves several centimeters past the target.

I think the simplest fallback solution would be to have the cover component periodically trigger a sensor update while the desk is in motion.

j5lien, 2021-07-25:
It supposed to be the case. When the bluetooth notification can't be setup, it supposed to retrieve the height when the desk moves.

On your log, what notification state can you see ?
```
[C][idasen_desk_controller:027]:   Notifications      : 
```

noxhirsch, 2021-07-25:
Notifications seem to be enabled:
`[15:20:59][C][idasen_desk_controller:026]:   Notifications      : enable`

Maybe the notification format is slightly different in some countries, so it can not get parsed? @spyfly seems to have the same problem and is also from Germany.

hoborm, 2021-08-15:
Clicking on the preset of the linak-desk-card sent the controller to the endless loop of "speed 0.0" and I could not interve from the physical controls either. However downloading and compiling a custom version with @noxhirsch's fix was the solution that can be found here: https://github.com/j5lien/esphome-idasen-desk-controller/pull/18 I downloaded his repository and referenced the component using this page as a guide: [https://esphome.io/components/external_components.html#external-components-local](https://esphome.io/components/external_components.html#external-components-local) . I believe I'm in the same situation as @angadsingh that I cannot use the ble_client version.

angadsingh, 2021-08-20:
thanks @hoborm. i had to do the same. 1.2.0 with @noxhirsch's patch is perfect - except no ble notifications so when moving desk manually HA doesn't know its height.

noxhirsch, 2021-08-20:
@hoborm & @angadsingh My workaround with the ble_client is to set the `update_interval` for the sensor to 5s (or faster if you want to "annoy" the desk more ;) ). That way, the manual controls are only "locked" for a few seconds - and additionally you always have the most up to date height, even if you move the desk manually.

angadsingh, 2021-08-20:
thanks @noxhirsch however the ble_client version doesn't even connect to the desk for some of us: https://github.com/j5lien/esphome-idasen-desk-controller/issues/29

j5lien, 2021-09-11:
I merged @noxhirsch's patch on the v1.2.0 and created the v1.2.1 with it.

JDFS404, 2023-07-24:
> @hoborm & @angadsingh My workaround with the ble_client is to set the `update_interval` for the height sensor to 5s (or faster if you want to "annoy" the desk more ;) ). That way, the manual controls are only "locked" for a few seconds - and additionally you always have the most up to date height, even if you move the desk manually.

Two years later... I had to upgrade my HA Core to HAOS to have it all integrated in 1 RPi instead of scattered throughout my Docker containers. One of the things was upgrading the ESP instance to 7.0 and using the latest controller.

It got stuck updating the height, but with this `update_interval` it's all smooth as before! Many thanks :)