j5lien/esphome-idasen-desk-controller #19: Interrupting movement with manual controller breaks

bahlo, 2021-05-10:
Great work, I love having my desk in Home Assistant (and HomeKit) and it works really good.

I have one issue: If I click the up or down arrow and then interrupt the movement with the manual controller, I cannot use the manual controller until I click the stop button.

bahlo, 2021-05-10:
Might be a duplicate of https://github.com/j5lien/esphome-idasen-desk-controller/issues/17, sorry if that is the case.

bahlo, 2021-05-10:
Here are some logs with `bluetooth_callback: false`. It receives `'Desk Moving': Sending state OFF`, so it should stop attempting to move, shouldn't it?

```
[16:58:47][D][sensor:099]: 'Desk Height': Sending state 43.60000 cm with 1 decimals of accuracy
[16:58:47][D][idasen_desk_controller:274]: Update Desk - Move from 43.6 to 52.0
[16:58:47][D][idasen_desk_controller:199]: Desk bluetooth data: height 44.3 - speed 62.0
[16:58:47][D][sensor:099]: 'Desk Height': Sending state 44.30000 cm with 1 decimals of accuracy
[16:58:47][D][idasen_desk_controller:274]: Update Desk - Move from 44.3 to 52.0
[16:58:47][D][idasen_desk_controller:199]: Desk bluetooth data: height 45.2 - speed 62.0
[16:58:47][D][sensor:099]: 'Desk Height': Sending state 45.20000 cm with 1 decimals of accuracy
[16:58:47][D][idasen_desk_controller:274]: Update Desk - Move from 45.2 to 52.0
[16:58:48][D][idasen_desk_controller:199]: Desk bluetooth data: height 45.9 - speed 62.0
[16:58:48][D][sensor:099]: 'Desk Height': Sending state 45.90000 cm with 1 decimals of accuracy
[16:58:48][D][idasen_desk_controller:274]: Update Desk - Move from 45.9 to 52.0
[16:58:48][D][idasen_desk_controller:199]: Desk bluetooth data: height 46.4 - speed 0.0
[16:58:48][D][sensor:099]: 'Desk Height': Sending state 46.40000 cm with 1 decimals of accuracy

// Move manual control

[16:58:48][D][binary_sensor:036]: 'Desk Moving': Sending state OFF
[16:58:48][D][idasen_desk_controller:274]: Update Desk - Move from 46.4 to 52.0
[16:58:48][D][idasen_desk_controller:199]: Desk bluetooth data: height 46.4 - speed 0.0
[16:58:48][D][idasen_desk_controller:274]: Update Desk - Move from 46.4 to 52.0
[16:58:48][D][idasen_desk_controller:199]: Desk bluetooth data: height 46.4 - speed 0.0
[16:58:48][D][idasen_desk_controller:274]: Update Desk - Move from 46.4 to 52.0
[16:58:48][D][idasen_desk_controller:199]: Desk bluetooth data: height 46.4 - speed 0.0
[16:58:48][D][idasen_desk_controller:274]: Update Desk - Move from 46.4 to 52.0
```

j5lien, 2021-05-10:
Thanks for this feedback, I haven't tested this kind of scenario. I won't be able to look at it for the moment but when I'll have time I'll try to fix that.

simonszu, 2021-05-27:
There is a similar issue: When i call `set_cover_position` to move the desk to a standing position, it moves to this position and stops. But apparently afterwards the desk is in a state to move to this position at all costs, so i cannot move it to a lower position with the physical switch. 

Current workaround is to call the `set_cover_position`, wait until the `desk_height` has the desired state, and then explicitly call `stop_cover` to "free" the desk from this state. Afterwards it is fine to move with the physical button again. 

Drawbacks of the workaround: I cannot stop movement of the desk when called via the ESP at all with the physical button. Also i fear that whenever the overload protection is triggered when the desk is rising, the desk will stop and be in a non-defined state by the state machine i am building with the workaround in nodered.

j5lien, 2021-06-19:
Is the new version based on ESPHome BLE Client fixes your issue ?

bahlo, 2021-06-20:
Sadly it's not fixing the issue. Instead with the new version everything takes _way_ longer if it works at all, so I reverted to 1.2.0.

[This is my configuration](https://github.com/bahlo/esphome/blob/idasen-desk-controller-2/office.yaml), I basically copied from the README.

j5lien, 2021-06-28:
The v2.0.1 release might fixes issues with the v2.0.0.

bahlo, 2021-06-28:
Okay so I tried running v2.0.1 and it actually crashed. First I moved it from 22 % to 24 % (via HASS) and I kept getting these `Move from 22 to 24` logs even though it stopped moving. Then I moved back to 22 % and it crashed:

```
[18:01:37][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:38][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:38][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:38][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:38][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:38][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:39][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:39][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:39][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:39][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:39][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:40][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:40][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:40][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:40][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:40][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:41][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:41][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:41][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:41][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:41][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:42][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:42][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:42][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:42][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:42][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:43][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:43][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:43][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:43][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:43][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:44][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:44][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:44][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:44][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:44][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:45][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:45][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:45][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:45][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:45][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 24
[18:01:46][D][cover:072]: 'Desk' - Setting
[18:01:46][D][cover:080]:   Position: 23%
[18:01:46][D][idasen_desk_controller:217]: Update Desk - Move from 22 to 23
[18:01:46][D][cover:072]: 'Desk' - Setting
[18:01:46][D][cover:080]:   Position: 22%
[18:01:52]Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
[18:01:52]Core 0 register dump:
[18:01:52]PC       0x4008df  PS       0x00060b4  A0       0x80090b1  A1       0x3fe760
[18:01:52]A2       0x3fbe8  A3       0x3fb3e  A4       0x00000000  A5       0x0000001a
[18:01:52]A6       0x4011c6f A7       0x3fc554  A8       0x3f0de  A9       0x3fe740
[18:01:52]A10      0x00000019  A11      0x00000000  A12      0x3fb124  A13      0x00000040
[18:01:52]A14      0x00000000  A15      0x3fb3d  SAR      0x0000001e EXCCAUSE 0x00000005
[18:01:52]EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff
[18:01:52]
[18:01:52]ELF file SHA256: 000000000000000
[18:01:52]
[18:01:52]Backtrace:0x4008df:0x3fe7600x40090be0x3fe790
[18:01:52]
[18:01:52]ore 1 register dump:
[18:01:52]PC       0x401e2e  PS       0x00060a4  A0       0x801b9a  A1       0x3fb6e
[18:01:52]A2       0x00000000  A3       0x00000001  A4       0x80090d0  A5       0x3fc330
[18:01:52]A6       0x00000000  A7       0x3fb3e  A8       0x00060023  A9       0x3fb7e
[18:01:52]A10      0x00000001  A11      0x00000001  A12      0x80090d0  A13      0x3fb5e
[18:01:52]A14      0x00000000  A15      0x3fb3e  SAR      0x00000000  EXCCAUSE 0x00000005
[18:01:52]EXCVADDR 0x00000000  LBEG     0x00000000  LEND     0x00000000  LCOUNT   0x00000000
[18:01:52]
[18:01:52]ELF file SHA256: 000000000000000
[18:01:52]
[18:01:52]Backtrace:0x401e2e:0x3fb6e 0x401b9a:0x3fb700 0x4009234d0x3fb720 0x40090be0x3fb740
[18:01:52]
[18:01:52]Rebooting...
```

j5lien, 2021-07-20:
On the 2.1.0, have you try to set the `only_up_down_command` option to `true` ?
https://github.com/j5lien/esphome-idasen-desk-controller#not-moving-using-cover-component

bahlo, 2021-07-21:
Just tried it, it's way more responsive than without it, but it doesn't stop at the correct height. If I interrupt with the controller I still have this issue.

j5lien, 2021-07-25:
When you interrupt with the controller, it normally takes 1 or 2 seconds to understand the desk stopped and correctly terminate the moving session.

I can't reproduce this issue where it doesn't stop to the correct height, so I don't known how to fix it.

You may try to change the value of the desk moving refresh from `200` to `100` or `50` https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/components/idasen_desk_controller/idasen_desk_controller.cpp#L20