j5lien/esphome-idasen-desk-controller #21: esphome desk controller not connecting

bernie-gitmans, 2021-05-23:
Thank you for all the effort you have put into developing this controller!

I run into an issue where the controller finds my desk, however it doesn't add it to be controlled.

This is a line from the log output:
'[0;35m[C][idasen_desk_controller:053]: BLE Device found: Name: Desk 1403, Address: fd:7f:6d:4a:a1:bb, serviceUUID: 99fa0001-338a-1024-8a49-009c0215f78a, txPower: 4[0m'

The controller just keeps scanning for BLE devices.
Could it be that the name: Desk 1403 is causing problems?

j5lien, 2021-05-23:
I don't think the name is an issue.
Could you share more logs ?

bernie-gitmans, 2021-05-23:
Of course:

`[16:29:47][C][wifi:319]:   DNS2: 0.0.0.0
[16:29:47][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=9858 (now=10122)
[16:29:47][C][logger:189]: Logger:
[16:29:47][C][logger:190]:   Level: VERY_VERBOSE
[16:29:47][C][logger:191]:   Log Baud Rate: 115200
[16:29:47][C][logger:192]:   Hardware UART: UART0
[16:29:47][C][ota:029]: Over-The-Air Updates:
[16:29:47][C][ota:030]:   Address: desk_controller.local:3232
[16:29:47][C][api:095]: API Server:
[16:29:47][C][api:096]:   Address: desk_controller.local:6053
[16:29:47][C][idasen_desk_controller:098]: Idasen Desk Controller:
[16:29:47][C][idasen_desk_controller:099]:   Mac address: 00:00:00:00:00:00
[16:29:47][C][idasen_desk_controller:100]:   Bluetooth callback: true
[16:29:47][C][idasen_desk_controller:101]:   Desk height 'Desk Height'
[16:29:47][C][idasen_desk_controller:101]:     Unit of Measurement: 'cm'
[16:29:47][C][idasen_desk_controller:101]:     Accuracy Decimals: 1
[16:29:47][C][idasen_desk_controller:101]:     Icon: 'mdi:arrow-up-down'
[16:29:47][C][idasen_desk_controller:102]:   Desk moving 'Desk Moving'
[16:29:47][C][idasen_desk_controller:102]:     Device Class: 'moving'
[16:29:47][C][idasen_desk_controller:103]:   Desk connection 'Desk Connection'
[16:29:47][C][idasen_desk_controller:103]:     Device Class: 'connectivity'
[16:29:47][C][idasen_desk_controller:104]:   Desk 'Desk'
[16:29:47][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=10058 (now=10276)
[16:29:48][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=10258 (now=10458)
[16:29:48][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=10458 (now=10661)
[16:29:48][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=10658 (now=10859)
[16:29:48][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=10858 (now=11058)
[16:29:48][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=11058 (now=11258)
[16:29:49][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=11258 (now=11458)
[16:29:49][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=11458 (now=11658)
[16:29:49][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=11658 (now=11858)
[16:29:49][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=11858 (now=12058)
[16:29:49][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=12058 (now=12258)
[16:29:50][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=12258 (now=12458)
[16:29:50][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=12458 (now=12658)
[16:29:50][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=12658 (now=12858)
[16:29:50][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=12858 (now=13058)
[16:29:50][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=13058 (now=13258)
[16:29:51][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=13258 (now=13458)
[16:29:51][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=13458 (now=13658)
[16:29:51][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=13658 (now=13858)
[16:29:51][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=13858 (now=14058)
[16:29:51][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=14058 (now=14258)
[16:29:52][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=14258 (now=14458)
[16:29:52][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=14458 (now=14658)
[16:29:52][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=14658 (now=14858)
[16:29:52][VV][api.service:220]: on_ping_request: PingRequest {}
[16:29:52][VV][api.service:032]: send_ping_response: PingResponse {}
[16:29:52][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=14858 (now=15058)
[16:29:52][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=15058 (now=15260)
[16:29:53][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=15258 (now=15458)
[16:29:53][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=15458 (now=15658)
[16:29:53][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=15658 (now=15858)
[16:29:53][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=15858 (now=16058)
[16:29:53][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=16058 (now=16258)
[16:29:54][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=16258 (now=16458)
[16:29:54][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=16458 (now=16658)
[16:29:54][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=16658 (now=16858)
[16:29:54][VV][scheduler:152]: Running interval 'update_desk' with interval=200 last_execution=16858 (now=17058)
[... cut ...]
[16:29:58][C][idasen_desk_controller:053]: BLE Device found: Name: , Address: 48:1b:7e:dc:13:42, manufacturer data: 4c00100509984f8908, txPower: 12
[16:29:59][C][idasen_desk_controller:053]: BLE Device found: Name: , Address: 6a:f0:a1:18:3e:03, manufacturer data: 4c000c0e00a978b990a83f112baad1fc70d8
[16:29:59][C][idasen_desk_controller:053]: BLE Device found: Name: , Address: 72:a3:9f:2d:ac:2a, manufacturer data: 4c001006271eb2c0c581, txPower: 8
[16:30:00][C][idasen_desk_controller:053]: BLE Device found: Name: , Address: 6d:63:01:b2:d9:bf, manufacturer data: 060001092002788a420a500719deeb902898e96d21cf14dcc8d2e1fade`

bernie-gitmans, 2021-05-23:
and in addition:

[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -61
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 04
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xa
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x0a (), length: 1, data: f8
[D][BLEAdvertisedDevice.cpp:478] setTXPower(): - txPower: -8
[V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 9, data: 4465736b2031343033
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: Desk 1403
[0;35m[C][idasen_desk_controller:053]: BLE Device found: Name: Desk 1403, Address: fd:7f:6d:4a:a1:bb, txPower: -8[0m
[V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!

j5lien, 2021-05-23:
This doesn't seem right : `[16:29:47][C][idasen_desk_controller:099]: Mac address: 00:00:00:00:00:00`
You need to set your desk mac address in the configuration
```yaml
idasen_desk_controller:
    mac_address: "FD:7F:6D:4A:A1:BB"
```

bernie-gitmans, 2021-05-24:
Yes, that did solve the problem.
I was thinking that the software would automatically implement the found mac_address.
Perhaps it is possible to put this in the readme?

closed 2021-06-04 by bernie-gitmans