j5lien/esphome-idasen-desk-controller #22: Crash after initial connection

SeraphimSerapis, 2021-05-25:
Hi all,
I'm running into this issue with my ESP32 controlling my desk. The setup is a [AZDelivery NodeMCU](https://www.az-delivery.de/en/products/esp32-developmentboard). I tried playing with build flags etc to resolve the issue but this keeps on happening. Any idea what may cause this behavior?

Cheers,
Tim

```
[10:53:55][C][idasen_desk_controller:053]: BLE Device found: Name: Tim Desk, Address: e9:43:60:ec:XX:XX, txPower: -8
[10:53:55][C][idasen_desk_controller:122]: Connecting client to device e9:43:60:ec:XX:XX
[10:53:57][C][idasen_desk_controller:126]: Fail to connect to client
[10:53:59][C][idasen_desk_controller:122]: Connecting client to device e9:43:60:ec:XX:XX
[10:53:59][D][binary_sensor:036]: 'Office ESP Desk Desk Connection': Sending state ON
[10:53:59][C][idasen_desk_controller:132]: Retrieve input remote characteristic.
[10:54:01][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[10:54:01][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[10:54:01][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[10:54:01][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[10:54:01][C][idasen_desk_controller:137]: Retrieve output remote characteristic.
[10:54:01][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[10:54:01][E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown
[10:54:01][C][idasen_desk_controller:142]: Register notification callback on output characteristic.
[10:54:01][C][idasen_desk_controller:148]Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
[10:54:01]Core 0 register dump:
[10:54:01]PC      : 0x801979c0  PS      : 0x00060930  A0      : 0x801738b2  A1      : 0x3ffec440
[10:54:01]A2      : 0x3ffec4a4  A3      : 0x3fff411c  A4      : 0x00060320  A5      : 0x00000000
[10:54:01]A6      : 0x00000001  A7      : 0x00000026  A8      : 0x801a32df  A9      : 0x3ffec3f0
[10:54:01]A10     : 0x3fff411c  A11     : 0x00000270  A12     : 0x00000000  A13     : 0x3fff4140
[10:54:01]A14     : 0x3fff40cc  A15     : 0x00000009  SAR     : 0x00000010  EXCCAUSE: 0x00000014
[10:54:01]EXCVADDR: 0x801979c0  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000
[10:54:01]
[10:54:01]Backtrace: 0x401979c0:0x3ffec440 0x401738af:0x3ffec460 0x40179a2f:0x3ffec480 0x401784e6:0x3ffec4a0 0x40198c05:0x3ffec4c0 0x4019620e:0x3ffec750 0x401979bd:0x3ffec770 0x40198c17:0x3ffeca00 0x4019620e:0x3ffecc90 0x401979bd:0x3ffeccb0 0x40198c17:0x3ffecf40 0x4019620e:0x3ffed1d0 0x401979bd:0x3ffed1f0 0x40198c17:0x3ffed480 0x4019620e:0x3ffed710 0x401979bd:0x3ffed730 0x40198f4b:0x3ffed9c0 0x40199133:0x3ffedc50 0x4019620e:0x3ffedc70 0x4019634d:0x3ffedc90 0x4019e92a:0x3ffedcb0 0x4017378a:0x3ffedcd0 0x4008e8c5:0x3ffedd00
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x401979c0: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x401738af: btu_free_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:504
WARNING Decoded 0x40179a2f: gatt_clcb_dealloc at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_utils.c:1761
WARNING Decoded 0x401784e6: GATTC_Write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/gatt/gatt_api.c:1040
WARNING Decoded 0x40198c05: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1163
WARNING Decoded 0x4019620e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x401979bd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x40198c17: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4019620e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x401979bd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x40198c17: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4019620e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x401979bd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x40198c17: bta_gattc_write at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1170
WARNING Decoded 0x4019620e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x401979bd: bta_gattc_pop_command_to_send at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1519
WARNING Decoded 0x40198f4b: bta_gattc_write_cmpl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1302
WARNING Decoded 0x40199133: bta_gattc_op_cmpl at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_act.c:1423
WARNING Decoded 0x4019620e: bta_gattc_sm_execute at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:288
WARNING Decoded 0x4019634d: bta_gattc_hdl_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/gatt/bta_gattc_main.c:397
WARNING Decoded 0x4019e92a: bta_sys_event at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/bta/sys/bta_sys_main.c:496
WARNING Decoded 0x4017378a: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:233
WARNING Decoded 0x4008e8c5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

j5lien, 2021-05-27:
Hello,

The crash seems to happen when the notification callback is registered.
Could you try to deactivate it ?

```yaml
idasen_desk_controller:
    mac_address: "00:00:00:00:00:00"
    bluetooth_callback: false

SeraphimSerapis, 2021-05-28:
Done! I got a little further and am now encountering the below issue. Amazon is delivering a Wemos D1 Mini ESP32 today and I'll see if that one works.

```
[11:46:56][D][binary_sensor:036]: 'Office ESP Desk Desk Connection': Sending state ON
[11:46:56][C][idasen_desk_controller:132]: Retrieve input remote characteristic.
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][C][idasen_desk_controller:137]: Retrieve output remote characteristic.
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][C][idasen_desk_controller:148]: Retrieve control remote characteristic.
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteCharacteristic.cpp:274] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[11:46:58][E][BLERemoteService.cpp:185] retrieveCharacteristics(): esp_ble_gattc_get_all_char: Unknown
[11:46:58][C][idasen_desk_controller:153]: Success connecting clabortbort(s callec lled atx40085449 on c7re
[11:46:58]
[11:46:58]Backtrace: 0x4009217c:0x3ffedb30 0x400923ad:0x3ffedb50 0x40085449:0x3ffedb70 0x40085474:0x3ffedb90 0x400855c5:0x3ffedbc0 0x4013127f:0x3ffedbe0 0x4012bc1d:0x3ffedea0 0x40129211:0x3ffedef0 0x401a32a6:0x3ffedf20 0x40173901:0x3ffedf40 0x40188825:0x3ffedf60 0x40180617:0x3ffedf80 0x40173614:0x3ffedfb0 0x4017387a:0x3ffedfd0 0x4008e8c5:0x3ffee000
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d ./.build/office_esp_desk/ -t idedata
WARNING Decoded 0x4009217c: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x400923ad: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x40085449: lock_init_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:81
WARNING Decoded 0x40085474: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:134
WARNING Decoded 0x400855c5: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
WARNING Decoded 0x4013127f: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
WARNING Decoded 0x4012bc1d: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
WARNING Decoded 0x40129211: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
WARNING Decoded 0x401a32a6: hash_map_set at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/osi/hash_map.c:130 (discriminator 1)
WARNING Decoded 0x40173901: btu_start_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:442
WARNING Decoded 0x40188825: smp_data_received at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/smp/smp_l2c.c:182
WARNING Decoded 0x40180617: l2c_rcv_acl_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c:274
WARNING Decoded 0x40173614: btu_hci_msg_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:137
WARNING Decoded 0x4017387a: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:229
WARNING Decoded 0x4008e8c5: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[11:47:00]
[11:47:00]Rebooting...
```

j5lien, 2021-06-19:
Is the new version based on ESPhome BLE Client fixes your issue ?

SeraphimSerapis, 2021-06-21:
No more crashes! What doesn't seem to work is the actual cover component, all the other sensors work great, though!

The cover says "moving" but nothing happens.

j5lien, 2021-06-21:
When you move the desk using the physical remote, does the sensors update automatically in real time ?

SeraphimSerapis, 2021-06-21:
Yes, that works flawlessly. It's just using the cover component that doesn't seem to work. My hunch is that my Linak controller may use different up and down commands.

j5lien, 2021-06-21:
The BLE client component should expose all services from the desk, you can look at the section "Setting Up Devices" of this page https://esphome.io/components/ble_client.html

Could share what you get ?

SeraphimSerapis, 2021-06-22:
This is what I get

```
[08:56:09][I][ble_client:083]: Attempting BLE connection to e9:43:60:ec:13:9a
[08:56:09][I][idasen_desk_controller:041]: [Desk] Connected successfully!
[08:56:09][I][ble_sensor:033]: [Desk Height] Connected successfully!
[08:56:09][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[08:56:09][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[08:56:12][I][ble_client:159]: Service UUID: 0x1800
[08:56:12][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[08:56:12][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[08:56:12][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[08:56:12][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[08:56:12][I][ble_client:159]: Service UUID: 0x1801
[08:56:12][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[08:56:12][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[08:56:12][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[08:56:12][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[08:56:12][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0xe, properties 0xc
[08:56:12][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0x12
[08:56:12][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[08:56:12][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[08:56:12][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x14, properties 0x1e
[08:56:12][I][ble_client:159]: Service UUID: 0x180A
[08:56:12][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[08:56:12][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[08:56:12][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[08:56:13][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[08:56:13][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[08:56:13][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0022338A10-248A-4900-9C02-15F78A, handle 0x20, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0023338A10-248A-4900-9C02-15F78A, handle 0x23, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0024338A10-248A-4900-9C02-15F78A, handle 0x26, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0025338A10-248A-4900-9C02-15F78A, handle 0x29, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0026338A10-248A-4900-9C02-15F78A, handle 0x2c, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0027338A10-248A-4900-9C02-15F78A, handle 0x2f, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0028338A10-248A-4900-9C02-15F78A, handle 0x32, properties 0x12
[08:56:13][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x35, properties 0x2
[08:56:13][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x37, properties 0x2
[08:56:13][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[08:56:13][I][ble_client:160]:   start_handle: 0x38  end_handle: 0xffff
[08:56:13][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x3a, properties 0xc
[08:56:13][I][ble_client:339]:  characteristic 99FA0032338A10-248A-4900-9C02-15F78A, handle 0x3c, properties 0xc
[08:56:13][I][ble_client:339]:  characteristic 99FA0033338A10-248A-4900-9C02-15F78A, handle 0x3e, properties 0xc
[08:56:13][I][ble_client:339]:  characteristic 99FA0034338A10-248A-4900-9C02-15F78A, handle 0x40, properties 0xc
[08:56:13][W][ble_client:170]: No descriptor found for notify of handle 0x1d
[08:56:13][W][ble_client:170]: No descriptor found for notify of handle 0x1d
```

j5lien, 2021-06-28:
OK the configuration is a bit different from my device:

```
[13:02:44][I][ble_client:159]: Service UUID: 0x1800
[13:02:44][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[13:02:44][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[13:02:44][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[13:02:44][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[13:02:44][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[13:02:44][I][ble_client:159]: Service UUID: 0x1801
[13:02:44][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[13:02:44][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[13:02:44][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[13:02:44][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[13:02:44][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[13:02:45][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[13:02:45][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[13:02:45][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[13:02:45][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[13:02:45][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[13:02:45][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[13:02:45][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[13:02:45][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[13:02:45][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[13:02:45][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[13:02:45][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[13:02:45][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
```

I added an option to use different control commands in the last release https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/README.md#not-moving-using-cover-component
Does it solve your issue ?

SeraphimSerapis, 2021-06-29:
I gave the new version a try and set `only_up_down_command: true` as instructed in the readme. Sadly no changes from the previous behaviour. Neither using the components up and down commands, nor the target position works.

j5lien, 2021-07-20:
Have you found other projects on github that allow you to control your desk and that work?

SeraphimSerapis, 2021-07-21:
I didn't get to try this myself, but apparently https://github.com/alex20465/deskbluez seems to be working for these desks. Sadly my job and a newborn child are currently reducing my "tinker time" to a bare minimum :(

j5lien, 2021-07-25:
With the option `only_up_down_command: true`, it supposed to use the same bluetooth controls as this project does.

> Sadly my job and a newborn child are currently reducing my "tinker time" to a bare minimum :(

I am in the same situation :-)

SeraphimSerapis, 2021-07-30:
Heads-up: I didn't change anything since the last time we spoke except disconnecting and reconnecting my ESP and: it works. No idea why, but it seems like my problem is resolved. Thanks again for your patience!

closed 2021-07-30 by SeraphimSerapis