j5lien/esphome-idasen-desk-controller #25: Desk Still not moving

Nicras, 2021-06-21:
Hi, and thank you a lot for this great work.
I struggled the whole day getting v2.0.0 up and running. Everything went fine except the desk is not moving when trying to move it from Homeassistant.

[image]

Then i saw you updated to v2.0.1 and I went straight updating my device but again the same behaviour.

Any help is appreciated

j5lien, 2021-06-21:
Could you share the logs from the start ?

Nicras, 2021-06-21:
i gues you mean this:

`[22:41:59][C][template.binary_sensor:018]: Template Binary Sensor 'Desk Connection'
[22:41:59][C][template.binary_sensor:018]: Template Binary Sensor 'Desk Moving'
[22:41:59][C][logger:189]: Logger:
[22:41:59][C][logger:190]:   Level: DEBUG
[22:41:59][C][logger:191]:   Log Baud Rate: 115200
[22:41:59][C][logger:192]:   Hardware UART: UART0
[22:41:59][C][esp32_ble_tracker:606]: BLE Tracker:
[22:41:59][C][esp32_ble_tracker:607]:   Scan Duration: 300 s
[22:41:59][C][esp32_ble_tracker:608]:   Scan Interval: 320.0 ms
[22:41:59][C][esp32_ble_tracker:609]:   Scan Window: 30.0 ms
[22:41:59][C][esp32_ble_tracker:610]:   Scan Type: ACTIVE
[22:41:59][C][ble_client:033]: BLE Client:
[22:41:59][C][ble_client:034]:   Address: e4:6b:42:7f:1a:39
[22:41:59][C][ble_sensor:019]: BLE Sensor 'Desk Height'
[22:41:59][C][ble_sensor:019]:   State Class: ''
[22:41:59][C][ble_sensor:019]:   Unit of Measurement: 'cm'
[22:41:59][C][ble_sensor:019]:   Accuracy Decimals: 1
[22:41:59][C][ble_sensor:019]:   Icon: 'mdi:arrow-up-down'
[22:41:59][C][ble_sensor:020]:   MAC address        : e4:6b:42:7f:1a:39
[22:42:00][C][ble_sensor:021]:   Service UUID       : 99FA0020338A10-248A-4900-9C02-15F78A
[22:42:00][C][ble_sensor:022]:   Characteristic UUID: 99FA0021338A10-248A-4900-9C02-15F78A
[22:42:00][C][ble_sensor:023]:   Descriptor UUID    : 00000000000000-0000-0000-0000-000000
[22:42:00][C][ble_sensor:024]:   Notifications      : YES
[22:42:00][C][ble_sensor:025]:   Update Interval: 4294967.5s
[22:42:00][C][ble_sensor:019]: BLE Sensor 'Desk Speed'
[22:42:00][C][ble_sensor:019]:   State Class: ''
[22:42:00][C][ble_sensor:019]:   Unit of Measurement: 'cm/min'
[22:42:00][C][ble_sensor:019]:   Accuracy Decimals: 0
[22:42:00][C][ble_sensor:019]:   Icon: 'mdi:speedometer'
[22:42:00][C][ble_sensor:020]:   MAC address        : e4:6b:42:7f:1a:39
[22:42:00][C][ble_sensor:021]:   Service UUID       : 99FA0020338A10-248A-4900-9C02-15F78A
[22:42:00][C][ble_sensor:022]:   Characteristic UUID: 99FA0021338A10-248A-4900-9C02-15F78A
[22:42:00][C][ble_sensor:023]:   Descriptor UUID    : 00000000000000-0000-0000-0000-000000
[22:42:00][C][ble_sensor:024]:   Notifications      : YES
[22:42:00][C][ble_sensor:025]:   Update Interval: 4294967.5s
[22:42:00][C][ota:029]: Over-The-Air Updates:
[22:42:00][C][ota:030]:   Address: linak-desk-control.mainka.home:3232
[22:42:00][C][api:095]: API Server:
[22:42:00][C][api:096]:   Address: linak-desk-control.mainka.home:6053
[22:42:00][C][idasen_desk_controller:024]: Idasen Desk Controller:
[22:42:00][C][idasen_desk_controller:025]:   MAC address        : e4:6b:42:7f:1a:39
[22:42:00][C][idasen_desk_controller:026]:   Notifications      : disable
[22:42:00][C][idasen_desk_controller:027]:   Desk 'Desk'
[22:42:00][I][ble_client:159]: Service UUID: 0x1800
[22:42:00][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[22:42:00][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[22:42:00][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[22:42:00][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[22:42:00][I][ble_client:159]: Service UUID: 0x1801
[22:42:00][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[22:42:00][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[22:42:00][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[22:42:00][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[22:42:01][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0xe, properties 0xc
[22:42:01][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0x12
[22:42:01][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[22:42:01][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[22:42:01][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x14, properties 0x1e
[22:42:01][I][ble_client:159]: Service UUID: 0x180A
[22:42:01][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[22:42:01][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[22:42:01][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[22:42:01][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[22:42:01][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[22:42:01][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0022338A10-248A-4900-9C02-15F78A, handle 0x20, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0023338A10-248A-4900-9C02-15F78A, handle 0x23, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0024338A10-248A-4900-9C02-15F78A, handle 0x26, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0025338A10-248A-4900-9C02-15F78A, handle 0x29, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0026338A10-248A-4900-9C02-15F78A, handle 0x2c, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0027338A10-248A-4900-9C02-15F78A, handle 0x2f, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0028338A10-248A-4900-9C02-15F78A, handle 0x32, properties 0x12
[22:42:01][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x35, properties 0x2
[... cut ...]
[22:42:01][I][ble_client:339]:  characteristic 99FA0032338A10-248A-4900-9C02-15F78A, handle 0x3c, properties 0xc
[22:42:01][I][ble_client:339]:  characteristic 99FA0033338A10-248A-4900-9C02-15F78A, handle 0x3e, properties 0xc
[22:42:01][I][ble_client:339]:  characteristic 99FA0034338A10-248A-4900-9C02-15F78A, handle 0x40, properties 0xc
[22:42:01][D][esp32_ble_tracker:180]: Starting scan...
[22:42:01][W][ble_client:170]: No descriptor found for notify of handle 0x1d
[22:42:01][W][ble_client:170]: No descriptor found for notify of handle 0x1d`

j5lien, 2021-06-28:
Your desk doesn't have exactly the same controls as mine.
On the latest release, I added an option to use different bluetooth control commands.
https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/README.md#not-moving-using-cover-component

Does it fix your issues ?

Nicras, 2021-06-28:
Hi, apparently not...since 2.0.1 I get the same disconnects as in [#26](https://github.com/j5lien/esphome-idasen-desk-controller/issues/26)
I figured out it connects correctly and if I move the table with the Linak Controller (which is [DPG1C](https://www.linak.de/geschaeftsbereiche/deskline/start/dpg1c/)), the following error message is logged and the ble_client disconnects

> [23:43:29][I][ble_client:083]: Attempting BLE connection to e4:6b:42:7f:1a:39
[23:43:29][V][ble_client:113]: [e4:6b:42:7f:1a:39] ESP_GATTC_OPEN_EVT
[23:43:29][I][idasen_desk_controller:042]: [] Connected successfully!
[23:43:29][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[23:43:33][V][ble_client:132]: cfg_mtu status 0, mtu 23
[23:43:33][V][ble_client:157]: [e4:6b:42:7f:1a:39] ESP_GATTC_SEARCH_CMPL_EVT
[23:43:33][I][ble_client:159]: Service UUID: 0x1800
[23:43:33][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[23:43:33][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[23:43:33][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[23:43:33][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[23:43:33][I][ble_client:159]: Service UUID: 0x1801
[23:43:33][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[23:43:33][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0xb
[23:43:33][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[23:43:33][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[23:43:33][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0xe, properties 0xc
[23:43:33][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x11
[23:43:33][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[23:43:33][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[23:43:33][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x14, properties 0x1e
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x15
[23:43:33][I][ble_client:159]: Service UUID: 0x180A
[23:43:33][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[23:43:33][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[23:43:33][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[23:43:33][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[23:43:33][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[23:43:33][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x1e
[23:43:33][I][ble_client:339]:  characteristic 99FA0022338A10-248A-4900-9C02-15F78A, handle 0x20, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x21
[23:43:33][I][ble_client:339]:  characteristic 99FA0023338A10-248A-4900-9C02-15F78A, handle 0x23, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x24
[23:43:33][I][ble_client:339]:  characteristic 99FA0024338A10-248A-4900-9C02-15F78A, handle 0x26, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x27
[23:43:33][I][ble_client:339]:  characteristic 99FA0025338A10-248A-4900-9C02-15F78A, handle 0x29, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x2a
[23:43:33][I][ble_client:339]:  characteristic 99FA0026338A10-248A-4900-9C02-15F78A, handle 0x2c, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x2d
[23:43:33][I][ble_client:339]:  characteristic 99FA0027338A10-248A-4900-9C02-15F78A, handle 0x2f, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x30
[23:43:33][I][ble_client:339]:  characteristic 99FA0028338A10-248A-4900-9C02-15F78A, handle 0x32, properties 0x12
[23:43:33][V][ble_client:374]:    descriptor 0x2902, handle 0x33
[23:43:33][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x35, properties 0x2
[23:43:33][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x37, properties 0x2
[23:43:33][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[23:43:33][I][ble_client:160]:   start_handle: 0x38  end_handle: 0xffff
[23:43:33][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x3a, properties 0xc
[23:43:33][I][ble_client:339]:  characteristic 99FA0032338A10-248A-4900-9C02-15F78A, handle 0x3c, properties 0xc
[23:43:33][I][ble_client:339]:  characteristic 99FA0033338A10-248A-4900-9C02-15F78A, handle 0x3e, properties 0xc
[23:43:33][I][ble_client:339]:  characteristic 99FA0034338A10-248A-4900-9C02-15F78A, handle 0x40, properties 0xc
[23:43:33][D][esp32_ble_tracker:180]: Starting scan...
[23:43:33][V][app:081]: A component took a long time in a loop() cycle (0.37 s).
[23:43:33][V][app:082]: Components should block for at most 20-30ms in loop().
[23:43:34][D][esp32_ble_tracker:620]: Found device 31:2B:11:4E:FE:92 RSSI=-82
[23:43:34][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[23:43:35][D][esp32_ble_tracker:620]: Found device 0E:7B:8A:77:E1:53 RSSI=-87
[23:43:35][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[23:43:37][D][esp32_ble_tracker:620]: Found device E4:6B:42:7F:1A:39 RSSI=-71
[... cut ...]
[23:43:49][D][esp32_ble_tracker:620]: Found device 4E:60:32:57:8B:C5 RSSI=-75
[23:43:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[23:43:49][D][esp32_ble_tracker:645]:   TX Power: 2
[23:44:04][V][ble_client:140]: [e4:6b:42:7f:1a:39] ESP_GATTC_DISCONNECT_EVT
[23:44:04][W][idasen_desk_controller:049]: [] Disconnected!
<

Maybe you can give me an advice how to figure out the right controls for my desk and where to change them in the source code?
Honestly I'm not a programmer but I'd really love to get this working, so any help would be appreciated.

j5lien, 2021-07-20:
I'm not sure to understand why you receive a `ESP_GATTC_DISCONNECT_EVT` from the desk.
If you don't configure this custom component and only try to retrieve the height sensor, does it work ?
If it works, the sensor is updated instantly or does it update every 30s ?

```yaml
esp32_ble_tracker:

ble_client:
  - mac_address: "00:00:00:00:00:00" # Replace with the desk bluetooth mac address
    id: idasen_desk

sensor:
  # Desk Height Sensor
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_height
    name: 'Desk Height'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:arrow-up-down'
    unit_of_measurement: 'cm'
    accuracy_decimals: 1
    update_interval: 30s
    notify: true
    lambda: |-
      uint16_t raw_height = ((uint16_t)x[1] << 8) | x[0];
      unsigned short height_mm = raw_height / 10;

      return (float) height_mm / 10;

j5lien, 2021-07-20:
Do you have another device connected to your desk with bluetooth (like your phone) ?

Nicras, 2021-07-29:
Heya, yeah I had/have my Phone connected, using the app. But I also tried the connection to ESP while my mobile was disconnected and I reset the desk controller, so it no other connection than ESP. I also tried the ble connection only method, you described above but it is still the same problem:

`[13:27:11][W][ble_sensor:039]: [Desk Height] Disconnected!
[13:27:11][V][sensor:035]: 'Desk Height': Received new state nan
[13:27:11][D][sensor:127]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[13:27:11][V][ble_client:113]: [e4:6b:42:7f:1a:39] ESP_GATTC_OPEN_EVT
[13:27:11][W][ble_client:115]: connect to e4:6b:42:7f:1a:39 failed, status=133
[13:27:11][D][esp32_ble_tracker:180]: Starting scan...
[13:27:12][D][ble_client:045]: Found device at MAC address [E4:6B:42:7F:1A:39]
[13:27:12][I][ble_client:083]: Attempting BLE connection to e4:6b:42:7f:1a:39
[13:27:12][V][ble_client:113]: [e4:6b:42:7f:1a:39] ESP_GATTC_OPEN_EVT
[13:27:12][I][ble_sensor:033]: [Desk Height] Connected successfully!
[13:27:15][V][ble_client:132]: cfg_mtu status 0, mtu 23
[13:27:15][V][ble_client:157]: [e4:6b:42:7f:1a:39] ESP_GATTC_SEARCH_CMPL_EVT
[13:27:15][I][ble_client:159]: Service UUID: 0x1800
[13:27:15][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[13:27:15][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[13:27:15][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[13:27:15][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[13:27:15][I][ble_client:159]: Service UUID: 0x1801
[13:27:15][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[13:27:15][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[13:27:15][V][ble_client:374]:    descriptor 0x2902, handle 0xb
[13:27:15][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[13:27:15][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[13:27:15][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0xe, properties 0xc
[13:27:15][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0x12
[13:27:15][V][ble_client:374]:    descriptor 0x2902, handle 0x11
[13:27:15][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[13:27:15][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[13:27:15][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x14, properties 0x1e
[13:27:15][V][ble_client:374]:    descriptor 0x2902, handle 0x15
[13:27:15][I][ble_client:159]: Service UUID: 0x180A
[13:27:15][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[13:27:15][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[13:27:15][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[13:27:16][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[13:27:16][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[13:27:16][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x1e
[13:27:16][I][ble_client:339]:  characteristic 99FA0022338A10-248A-4900-9C02-15F78A, handle 0x20, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x21
[13:27:16][I][ble_client:339]:  characteristic 99FA0023338A10-248A-4900-9C02-15F78A, handle 0x23, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x24
[13:27:16][I][ble_client:339]:  characteristic 99FA0024338A10-248A-4900-9C02-15F78A, handle 0x26, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x27
[13:27:16][I][ble_client:339]:  characteristic 99FA0025338A10-248A-4900-9C02-15F78A, handle 0x29, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x2a
[13:27:16][I][ble_client:339]:  characteristic 99FA0026338A10-248A-4900-9C02-15F78A, handle 0x2c, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x2d
[13:27:16][I][ble_client:339]:  characteristic 99FA0027338A10-248A-4900-9C02-15F78A, handle 0x2f, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x30
[13:27:16][I][ble_client:339]:  characteristic 99FA0028338A10-248A-4900-9C02-15F78A, handle 0x32, properties 0x12
[13:27:16][V][ble_client:374]:    descriptor 0x2902, handle 0x33
[13:27:16][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x35, properties 0x2
[13:27:16][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x37, properties 0x2
[13:27:16][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[13:27:16][I][ble_client:160]:   start_handle: 0x38  end_handle: 0xffff
[13:27:16][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x3a, properties 0xc
[13:27:16][I][ble_client:339]:  characteristic 99FA0032338A10-248A-4900-9C02-15F78A, handle 0x3c, properties 0xc
[13:27:16][I][ble_client:339]:  characteristic 99FA0033338A10-248A-4900-9C02-15F78A, handle 0x3e, properties 0xc
[13:27:16][I][ble_client:339]:  characteristic 99FA0034338A10-248A-4900-9C02-15F78A, handle 0x40, properties 0xc
[13:27:16][D][esp32_ble_tracker:180]: Starting scan...
[13:27:16][V][app:081]: A component took a long time in a loop() cycle (0.37 s).
[13:27:16][V][app:082]: Components should block for at most 20-30ms in loop().
[... cut ...]
[13:27:29][D][esp32_ble_tracker:620]: Found device 19:78:64:CF:15:4F RSSI=-76
[13:27:29][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:27:35][D][esp32_ble_tracker:620]: Found device 24:4B:03:D6:15:3E RSSI=-94
[13:27:35][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[13:27:44][D][esp32_ble_tracker:620]: Found device 37:47:55:02:83:EB RSSI=-83
[13:27:44][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:27:47][V][ble_client:140]: [e4:6b:42:7f:1a:39] ESP_GATTC_DISCONNECT_EVT
[13:27:47][W][ble_sensor:039]: [Desk Height] Disconnected!`

boardwarp, 2021-12-17:
I've tried to follow the same diagnostic/replication steps with different configuration advice I've found, but they seem to result in the same behaviour. The whole time I've got my phone in flight mode, so it's not trying to connect at the same time.

* After the firmware update is complete, put the desk in pairing mode and wait for it to pair with the ESP32
* Wait for a few minutes to see whether it disconnects with no interaction
  * If it does, reconnect by putting desk in pairing mode again
  * If a disconnection occurs at any point during the following steps, also re-pair the desk before proceeding
* Check values of sensors without interacting with desk
* Change desk height manually
* Change desk height using integration

#### Iteration 1
Using the basic configuration from the README, the cover appears in Home Assistant, but incorrectly reports 100% position. The desk disconnected by itself after a minute or two. It can be reconnected by putting the desk in pairing mode again. Attempting to change the desk height via Home Assistant in the interval between connection and disconnection starts looping as if it's working, but the height doesn't change. Attempting to raise/lower the desk via the cover entity does nothing.

```
esphome:
  name: desk-controller

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "a1d19bdd07f3866190e9b75d212b25c6"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.11
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Desk-Controller Fallback Hotspot"
    password: !secret fallback_password

captive_portal:

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v3.0.0

esp32_ble_tracker:

ble_client:
  - mac_address: "d6:ac:20:b7:c5:19"
    id: idasen_desk

idasen_desk_controller:
    ble_client_id: idasen_desk
    # Fallback to use only up and down commands (less precise)
    # -----------
    # Optional
    only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "Desk"
```

```
INFO Reading configuration /config/desk-controller.yaml...
INFO Starting log output from 192.168.0.11 using esphome API
INFO Successfully connected to 192.168.0.11
[... 1041 lines cut ...]
[12:51:49][D][esp32_ble_tracker:719]: Found device 76:E1:AE:73:72:C4 RSSI=-78
[12:51:49][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:51:49][D][esp32_ble_tracker:744]:   TX Power: 2
[12:51:49][D][esp32_ble_tracker:719]: Found device D8:0F:99:60:D1:78 RSSI=-82
[12:51:49][D][esp32_ble_tracker:740]:   Address Type: PUBLIC
[12:51:50][D][esp32_ble_tracker:719]: Found device 35:3E:9D:11:7A:F7 RSSI=-90
[12:51:50][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:51:52][D][esp32_ble_tracker:719]: Found device DB:A6:0E:85:0E:C7 RSSI=-90
[12:51:52][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:51:52][D][esp32_ble_tracker:719]: Found device 0F:AD:5B:79:85:B1 RSSI=-85
[12:51:52][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:51:56][D][esp32_ble_tracker:719]: Found device 5C:87:C4:FD:9E:57 RSSI=-68
[12:51:56][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:52:01][D][esp32_ble_tracker:719]: Found device 67:13:C5:49:81:D4 RSSI=-81
[12:52:01][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[12:52:01][D][esp32_ble_tracker:744]:   TX Power: 2
[12:52:04][D][esp32_ble_tracker:719]: Found device D0:03:4B:52:9C:4E RSSI=-83
[12:52:04][D][esp32_ble_tracker:740]:   Address Type: PUBLIC
[12:52:04][D][esp32_ble_tracker:744]:   TX Power: 2
[12:52:05][W][ble_sensor:082]: Error reading char at handle 29, status=5
[12:52:11][W][ble_sensor:039]: [Desk Height] Disconnected!
[12:52:11][D][sensor:113]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[12:52:11][D][ble_client:047]: Found device at MAC address [D6:AC:20:B7:C5:19]
[12:52:11][I][ble_client:085]: Attempting BLE connection to d6:ac:20:b7:c5:19
```

backslashV, 2022-03-02:
Any updates on this? Facing the same issue. Tried with two different ESP32 devices.