j5lien/esphome-idasen-desk-controller #26: Desk not moving / Disconnecting shortly after connection

Kamaroth92, 2021-06-24:
Not sure if this the same issue as #25 or not but it looks the ESP32 connects briefly to the desk but drops connection soon after. While connected I am unable to move the desk.
Running controller v2.0.1 with ESPHome v1.19.4

Connection issue:
```
[08:41:05][D][esp32_ble_tracker:180]: Starting scan...
[08:41:05][D][ble_client:045]: Found device at MAC address [C3:E7:62:F2:33:E6]
[08:41:05][I][ble_client:083]: Attempting BLE connection to c3:e7:62:f2:33:e6
[08:41:06][I][idasen_desk_controller:042]: [Computer Desk] Connected successfully!
[08:41:06][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[08:41:07][I][ble_client:159]: Service UUID: 0x1800
[08:41:07][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[08:41:07][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[08:41:07][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[08:41:07][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[08:41:07][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[08:41:07][I][ble_client:159]: Service UUID: 0x1801
[08:41:07][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[08:41:07][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[08:41:07][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[08:41:07][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[08:41:07][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[08:41:07][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[08:41:07][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[08:41:07][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[08:41:07][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[08:41:07][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[08:41:07][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[08:41:07][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[08:41:07][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[08:41:07][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[08:41:07][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[08:41:07][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[08:41:07][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
[08:41:07][D][esp32_ble_tracker:180]: Starting scan...
[08:41:08][D][esp32_ble_tracker:620]: Found device 52:60:99:8F:8C:A1 RSSI=-69
[08:41:08][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:08][D][esp32_ble_tracker:620]: Found device C3:E7:62:F2:33:E6 RSSI=-49
[08:41:08][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:08][D][esp32_ble_tracker:645]:   TX Power: 2
[08:41:09][D][esp32_ble_tracker:620]: Found device E7:E7:F2:2E:42:DC RSSI=-75
[08:41:09][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:09][D][esp32_ble_tracker:643]:   Name: '5852A70F0B4633A3E9'
[08:41:09][D][esp32_ble_tracker:645]:   TX Power: 2
[08:41:10][D][esp32_ble_tracker:620]: Found device 49:56:41:DB:0A:24 RSSI=-86
[08:41:10][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:10][D][esp32_ble_tracker:645]:   TX Power: 2
[08:41:10][D][esp32_ble_tracker:620]: Found device 52:F5:96:20:25:CD RSSI=-67
[08:41:10][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:10][D][esp32_ble_tracker:620]: Found device 04:FC:5A:D6:AA:9A RSSI=-46
[08:41:10][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:12][D][esp32_ble_tracker:620]: Found device C0:28:8D:F2:69:39 RSSI=-83
[08:41:12][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[08:41:12][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[08:41:15][D][esp32_ble_tracker:620]: Found device 5B:F9:28:B1:A3:37 RSSI=-85
[08:41:15][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:15][D][esp32_ble_tracker:620]: Found device 74:AC:B9:5C:E0:93 RSSI=-81
[08:41:15][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[08:41:31][D][esp32_ble_tracker:620]: Found device 09:10:78:F7:25:9B RSSI=-45
[08:41:31][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[08:41:38][W][idasen_desk_controller:049]: [Computer Desk] Disconnected!
[08:41:38][D][binary_sensor:036]: 'Desk Connection': Sending state OFF
```

Moving issue:
```
[08:42:17][D][ble_client:045]: Found device at MAC address [C3:E7:62:F2:33:E6]
[08:42:17][I][ble_client:083]: Attempting BLE connection to c3:e7:62:f2:33:e6
[08:42:17][I][idasen_desk_controller:042]: [Computer Desk] Connected successfully!
[08:42:17][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[... 55 lines cut ...]
[08:42:25][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:27][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 0
[08:42:27][D][cover:072]: 'Computer Desk' - Setting
[08:42:27][D][cover:076]:   Command: STOP
[08:42:27][D][idasen_desk_controller:248]: Cover control - STOP
[08:42:27][D][cover:072]: 'Computer Desk' - Setting
[08:42:27][D][cover:080]:   Position: 72%
[08:42:27][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 72
[08:42:27][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 72
[08:42:28][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 72
[08:42:28][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 72
[08:42:28][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 72
[08:42:28][D][cover:072]: 'Computer Desk' - Setting
[08:42:28][D][cover:076]:   Command: STOP
[08:42:28][D][idasen_desk_controller:248]: Cover control - STOP
[08:42:50][W][idasen_desk_controller:049]: [Computer Desk] Disconnected!
[08:42:50][D][binary_sensor:036]: 'Desk Connection': Sending state OFF
```

I can't reconnect the desk without putting it back into pairing mode.

j5lien, 2021-06-28:
I haven't encountered any connection problem since version 2.0.0 and using the ble client.

For the moving issue, on the latest release, I added an option to use different bluetooth control commands.
https://github.com/j5lien/esphome-idasen-desk-controller/blob/main/README.md#not-moving-using-cover-component

Does it fix your issues ?

eblechschmidt, 2021-06-30:
I'am not sure if it is the same issue but I also get the same warning as above but still in the phase when I try to pair the device.

```
[13:45:43][D][ble_client:045]: Found device at MAC address [EA:2B:E8:5A:DB:81]
[13:45:43][I][ble_client:083]: Attempting BLE connection to ea:2b:e8:5a:db:81
[13:45:44][I][ble_sensor:033]: [Desk Height] Connected successfully!
[13:45:44][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[13:45:44][I][idasen_desk_controller:042]: [] Connected successfully!
[13:45:44][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[13:45:45][I][ble_client:159]: Service UUID: 0x1800
[13:45:45][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[13:45:45][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[13:45:45][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[13:45:45][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[13:45:45][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[13:45:45][I][ble_client:159]: Service UUID: 0x1801
[13:45:45][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[13:45:45][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[13:45:45][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[13:45:45][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[13:45:45][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[13:45:45][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[13:45:45][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[13:45:45][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[13:45:45][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[13:45:45][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[13:45:45][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[13:45:45][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[13:45:45][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[13:45:46][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[13:45:46][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[13:45:46][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[13:45:46][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
[13:45:46][D][esp32_ble_tracker:180]: Starting scan...
[13:45:46][W][ble_client:170]: No descriptor found for notify of handle 0x1a
[13:45:46][W][ble_client:170]: No descriptor found for notify of handle 0x1a
[13:45:46][D][esp32_ble_tracker:620]: Found device EA:2B:E8:5A:DB:81 RSSI=-72
[13:45:46][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:45:46][D][esp32_ble_tracker:645]:   TX Power: 2
[13:45:46][D][esp32_ble_tracker:620]: Found device 19:9C:C9:66:8F:5E RSSI=-68
[13:45:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:45:48][D][esp32_ble_tracker:620]: Found device 72:44:5B:94:C6:C4 RSSI=-54
[13:45:48][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:45:48][D][esp32_ble_tracker:645]:   TX Power: 2
[13:45:49][D][esp32_ble_tracker:620]: Found device 16:BD:1B:7C:5C:4E RSSI=-59
[13:45:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[13:45:49][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:49][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:49][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[13:45:49][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:49][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:49][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[13:45:51][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:51][W][ble_sensor:082]: Error reading char at handle 26, status=5
[13:45:51][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[13:46:16][W][ble_sensor:039]: [Desk Height] Disconnected!
[13:46:16][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[13:46:16][W][ble_sensor:039]: [Desk Speed] Disconnected!
[13:46:16][D][sensor:117]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
[13:46:16][W][idasen_desk_controller:049]: [] Disconnected!
[13:46:16][D][binary_sensor:036]: 'Desk Connection': Sending state OFF
[13:46:20][D][ble_client:045]: Found device at MAC address [EA:2B:E8:5A:DB:81]
[13:46:20][I][ble_client:083]: Attempting BLE connection to ea:2b:e8:5a:db:81
```

j5lien, 2021-07-20:
Do you have another device connected to your desk with bluetooth ? (like a phone)

Could you check if you can retrieve the desk height without using this custom component like I explained here ? https://github.com/j5lien/esphome-idasen-desk-controller/issues/25#issuecomment-883697499

MunkeyBalls, 2021-08-09:
@j5lien , thanks for the project, would be cool to control my desk from HA

I'm running into the same issue as @Kamaroth92. I've disconnected my phone from the desk and changed the code you referenced to in your previous comment.

```
[12:33:05][I][ble_client:083]: Attempting BLE connection to XX:XX:XX:XX:XX
[12:33:05][W][idasen_desk_controller:049]: [Desk] Disconnected!
[12:33:05][W][ble_sensor:039]: [Desk Height] Disconnected!
[12:33:05][D][sensor:127]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[12:33:05][W][ble_client:115]: connect to XX:XX:XX:XX:XX failed, status=133
[12:33:06][D][ble_client:045]: Found device at MAC address [XX:XX:XX:XX:XX]
[12:33:06][I][ble_client:083]: Attempting BLE connection to XX:XX:XX:XX:XX
[12:33:06][I][idasen_desk_controller:042]: [Desk] Connected successfully!
[12:33:06][I][ble_sensor:033]: [Desk Height] Connected successfully!
[12:33:07][I][ble_client:159]: Service UUID: 0x1800
[12:33:07][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[12:33:07][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[12:33:07][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[12:33:07][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[12:33:07][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[12:33:07][I][ble_client:159]: Service UUID: 0x1801
[12:33:07][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[12:33:07][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[12:33:07][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[12:33:07][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[12:33:07][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[12:33:07][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[12:33:07][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[12:33:07][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[12:33:07][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[12:33:07][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[12:33:07][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[12:33:07][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[12:33:07][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[12:33:07][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[12:33:07][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[12:33:07][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[12:33:07][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
[12:33:07][W][ble_client:170]: No descriptor found for notify of handle 0x1a
[12:33:13][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[12:33:13][W][ble_sensor:082]: Error reading char at handle 26, status=5
[12:33:27][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[12:33:27][W][ble_sensor:082]: Error reading char at handle 26, status=5
[12:33:38][W][idasen_desk_controller:049]: [Desk] Disconnected!
[12:33:38][W][ble_sensor:039]: [Desk Height] Disconnected!
[12:33:38][D][sensor:127]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[12:33:40][D][ble_client:045]: Found device at MAC address [XX:XX:XX:XX:XX]
[12:33:40][I][ble_client:083]: Attempting BLE connection to XX:XX:XX:XX:XX
[12:33:56][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[12:34:10][W][idasen_desk_controller:049]: [Desk] Disconnected!
[12:34:10][W][ble_sensor:039]: [Desk Height] Disconnected!
[12:34:10][D][sensor:127]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[12:34:10][W][ble_client:115]: connect to XX:XX:XX:XX:XX failed, status=133
[12:34:10][D][esp32_ble_tracker:180]: Starting scan...
[12:34:10][D][ble_client:045]: Found device at MAC address [XX:XX:XX:XX:XX]
[12:34:10][I][ble_client:083]: Attempting BLE connection to XX:XX:XX:XX:XX
[12:34:10][W][idasen_desk_controller:049]: [Desk] Disconnected!
[12:34:10][W][ble_sensor:039]: [Desk Height] Disconnected!
[12:34:10][D][sensor:127]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[12:34:10][W][ble_client:115]: connect to XX:XX:XX:XX:XX failed, status=133
```

The esp32 will keep failing to connect until I hold the button, then it connects, throws some errors around and disconnects. After this is will continue to try and fail to reconnect until I hold the button.

Let me know if I can give you any further info.
```
Espressif ESP-WROVER-KIT
by espressif
Firmware: esphome v1.20.4 Aug 9 2021, 12:14:23 
```

hco, 2021-08-20:
I have the same behavior as MunkeyBalls, I think.

Morcains, 2021-08-23:
I have the same behavior, tried with both a Adafruit Feather ESP32 and a NoceMCU ESP32.

zackpollard, 2021-11-20:
Also experiencing this with the latest v3.0.0 of the component and ESPHome 2021.11.1 and 2021.10.3

backslashV, 2022-01-09:
Same issue on esphome v2021.12.1