j5lien/esphome-idasen-desk-controller #29: Connecting but not showing any parameters and not moving

sliwma, 2021-07-14:
Hey, I copy-pasted the config and it doesn't work - logs/config below.
Also, ESP connects to the desk only after holding the BT button on the controller - phone app can reconnect after disconnecting.

`
esphome:
  name: gateway-desk
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  manual_ip:
    static_ip: 192.168.12.43
    gateway: 192.168.12.1
    subnet: 255.255.255.0
  fast_connect: true
  ap:
    ssid: "Gateway-desk"
    password: !secret fallback

captive_portal:

logger:

api:
  password: !secret api_gateway_desk

ota:
  password: !secret api_gateway_desk

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v2.1.0

esp32_ble_tracker:

idasen_desk_controller:
    ble_client_id: idasen_desk
    only_up_down_command: true

globals:
  - id: ble_client_connected
    type: bool
    initial_value: 'false'

ble_client:
  - mac_address: "ED:D3:5C:57:32:0A"
    id: idasen_desk
    on_connect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = true;
        - delay: 5s
        - lambda: |-
            id(desk_height).update();
            id(desk_speed).update();
    on_disconnect:
      then:
        - lambda: |-
            id(ble_client_connected) = false;

cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  - platform: ble_client
    ble_client_id: idasen_desk
[... 198 lines cut ...]
[13:54:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 22
[13:54:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 22
[13:54:26][D][cover:072]: 'Desk' - Setting
[13:54:26][D][cover:080]:   Position: 71%
[13:54:26][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
...107 attempts...
[13:54:48][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:48][W][ble_sensor:039]: [Desk Height] Disconnected!
[13:54:48][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[13:54:48][W][ble_sensor:039]: [Desk Speed] Disconnected!
[13:54:48][D][sensor:117]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
[13:54:48][W][idasen_desk_controller:049]: [Desk] Disconnected!
[13:54:48][D][binary_sensor:036]: 'Desk Connection': Sending state OFF
[13:54:48][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:48][W][idasen_desk_controller:146]: [Desk] Error sending write request for cover, status=-1
[13:54:48][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:48][W][idasen_desk_controller:146]: [Desk] Error sending write request for cover, status=-1
[13:54:48][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:48][W][idasen_desk_controller:146]: [Desk] Error sending write request for cover, status=-1
[13:54:48][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[13:54:48][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[13:54:48][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:48][W][idasen_desk_controller:146]: [Desk] Error sending write request for cover, status=-1
[13:54:49][D][idasen_desk_controller:217]: Update Desk - Move from 100 to 71
[13:54:49][W][idasen_desk_controller:146]: [Desk] Error sending write request for cover, status=-1`

j5lien, 2021-07-20:
I don't understand why you need to hold the BT button on the controller to reconnect, it should be necessary only the first time.

On your logs, it seems to be an issue when retrieving the desk position and it could explain why it doesn't work as expected and it only accepts to go in one direction.
Could you check if you can retrieve the desk height without using this custom component like I explained here ? https://github.com/j5lien/esphome-idasen-desk-controller/issues/25#issuecomment-883697499

There might be an issue if your phone is also paired to the desk. I never paired mine to the desk and I don't know how the desk controller and the ESP32 handle this (not correctly I think).

sliwma, 2021-07-21:
Yeah, however:
```
[13:54:15][W][ble_sensor:039]: [Desk Speed] Disconnected!
[13:54:15][D][sensor:117]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
[13:54:15][W][idasen_desk_controller:049]: [Desk] Disconnected!
[13:54:15][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
[13:54:15][D][esp32_ble_tracker:180]: Starting scan...
[13:54:15][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[13:54:15][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[13:54:15][I][ble_sensor:033]: [Desk Height] Connected successfully!
[13:54:15][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[13:54:15][I][idasen_desk_controller:042]: [Desk] Connected successfully!
```
That's the part where I hold the button - I have paired the desk before.

Now the part after changes from another issue:
```
[19:58:10][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[19:58:23][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:58:23][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:58:23][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
[19:58:23][D][esp32_ble_tracker:180]: Starting scan...
[19:58:25][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[19:58:25][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[19:58:25][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:58:25][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:58:25][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
```
after holding the button:
```
[19:58:30][D][esp32_ble_tracker:180]: Starting scan...
[19:58:32][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[19:58:32][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[19:58:32][I][ble_sensor:033]: [Desk Height] Connected successfully!
[19:58:34][I][ble_client:159]: Service UUID: 0x1800
[19:58:34][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[19:58:34][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[19:58:34][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[19:58:34][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[19:58:34][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[19:58:34][I][ble_client:159]: Service UUID: 0x1801
[19:58:34][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[19:58:34][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[19:58:34][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[19:58:34][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[19:58:34][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[19:58:34][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[19:58:34][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[19:58:34][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[19:58:34][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[19:58:34][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[19:58:34][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[19:58:34][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[19:58:34][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[19:58:34][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[19:58:34][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[19:58:34][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[19:58:34][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
```
Not doing anything, happend by itself:
```
[19:59:05][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:59:05][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:59:05][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[19:59:05][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[19:59:10][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[19:59:35][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:59:35][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:59:35][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
[19:59:35][D][esp32_ble_tracker:180]: Starting scan...
[19:59:35][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[19:59:35][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[19:59:35][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:59:35][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:59:36][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
[19:59:36][D][esp32_ble_tracker:180]: Starting scan...
[19:59:36][D][ble_client:045]: Found device at MAC address [ED:D3:5C:57:32:0A]
[19:59:36][I][ble_client:083]: Attempting BLE connection to ed:d3:5c:57:32:0a
[19:59:36][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:59:36][D][sensor:117]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:59:36][W][ble_client:115]: connect to ed:d3:5c:57:32:0a failed, status=133
```

rcambrj, 2021-07-30:
I'm afraid that I have a very similar problem 😞 

```
INFO Reading configuration /config/desk_rob.yaml...
INFO Starting log output from desk_rob.local using esphome API
INFO Connecting to desk_rob.local:6053 (192.168.88.107)
INFO Successfully connected to desk_rob.local
[20:07:08][I][app:105]: ESPHome version 1.20.1 compiled on Jul 30 2021, 20:02:48
[20:07:08][C][wifi:499]: WiFi:
[20:07:08][C][wifi:359]:   SSID: [redacted]
[20:07:08][C][wifi:360]:   IP Address: 192.168.88.107
[20:07:08][C][wifi:362]:   BSSID: [redacted]
[20:07:08][C][wifi:363]:   Hostname: 'desk_rob'
[20:07:08][C][wifi:367]:   Signal strength: -73 dB ▂▄▆█
[20:07:08][C][wifi:371]:   Channel: 9
[20:07:08][C][wifi:372]:   Subnet: 255.255.255.0
[20:07:08][C][wifi:373]:   Gateway: 192.168.88.1
[20:07:08][C][wifi:374]:   DNS1: 192.168.88.1
[20:07:08][C][wifi:375]:   DNS2: 0.0.0.0
[20:07:08][C][logger:189]: Logger:
[20:07:08][C][logger:190]:   Level: DEBUG
[20:07:08][C][logger:191]:   Log Baud Rate: 115200
[20:07:08][C][logger:192]:   Hardware UART: UART0
[20:07:08][C][esp32_ble_tracker:606]: BLE Tracker:
[20:07:08][C][esp32_ble_tracker:607]:   Scan Duration: 300 s
[20:07:08][C][esp32_ble_tracker:608]:   Scan Interval: 320.0 ms
[20:07:08][C][esp32_ble_tracker:609]:   Scan Window: 30.0 ms
[20:07:08][C][esp32_ble_tracker:610]:   Scan Type: ACTIVE
[20:07:08][C][ble_client:033]: BLE Client:
[20:07:08][C][ble_client:034]:   Address: e3:b7:6e:c0:90:78
[20:07:08][C][ota:029]: Over-The-Air Updates:
[20:07:08][C][ota:030]:   Address: desk_rob.local:3232
[20:07:08][C][ota:032]:   Using Password.
[20:07:08][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[20:07:08][C][api:095]: API Server:
[20:07:08][C][api:096]:   Address: desk_rob.local:6053
[20:07:08][C][idasen_desk_controller:024]: Idasen Desk Controller:
[20:07:08][C][idasen_desk_controller:025]:   MAC address        : e3:b7:6e:c0:90:78
[20:07:08][C][idasen_desk_controller:026]:   Notifications      : disable
[20:07:08][C][idasen_desk_controller:027]:   Desk 'Rob's Desk'
[20:07:27][W][idasen_desk_controller:049]: [Rob's Desk] Disconnected!
[20:07:27][W][ble_client:115]: connect to e3:b7:6e:c0:90:78 failed, status=133
[20:07:27][D][esp32_ble_tracker:180]: Starting scan...
[20:07:27][D][ble_client:045]: Found device at MAC address [E3:B7:6E:C0:90:78]
[20:07:27][I][ble_client:083]: Attempting BLE connection to e3:b7:6e:c0:90:78
[20:07:27][W][idasen_desk_controller:049]: [Rob's Desk] Disconnected!
[20:07:27][W][ble_client:115]: connect to e3:b7:6e:c0:90:78 failed, status=133
[20:07:27][D][esp32_ble_tracker:180]: Starting scan...
```
...log shortened. at this point I press and hold the bluetooth button on the desk and the blue light starts flashing...
```
[20:07:37][D][esp32_ble_tracker:180]: Starting scan...
[20:07:37][D][ble_client:045]: Found device at MAC address [E3:B7:6E:C0:90:78]
[20:07:37][I][ble_client:083]: Attempting BLE connection to e3:b7:6e:c0:90:78
[20:07:37][I][idasen_desk_controller:042]: [Rob's Desk] Connected successfully!
[20:07:39][I][ble_client:159]: Service UUID: 0x1800
[20:07:39][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[20:07:39][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[20:07:39][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[20:07:39][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[20:07:39][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[20:07:39][I][ble_client:159]: Service UUID: 0x1801
[20:07:39][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[20:07:39][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[20:07:39][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[20:07:39][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[20:07:39][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[20:07:39][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[20:07:39][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[20:07:39][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[... 80 lines cut ...]
[20:08:43][D][esp32_ble_tracker:620]: Found device 07:32:F0:86:B7:E0 RSSI=-66
[20:08:43][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:43][D][esp32_ble_tracker:620]: Found device 6F:A9:0A:8D:70:DA RSSI=-70
[20:08:43][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:43][D][esp32_ble_tracker:645]:   TX Power: 2
[20:08:44][D][esp32_ble_tracker:620]: Found device 52:85:4C:84:5B:2B RSSI=-82
[20:08:44][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:44][D][esp32_ble_tracker:645]:   TX Power: 2
[20:08:44][D][esp32_ble_tracker:620]: Found device 00:42:79:E9:A5:1A RSSI=-65
[20:08:44][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[20:08:45][D][esp32_ble_tracker:620]: Found device 66:C7:DA:C6:1E:75 RSSI=-59
[20:08:45][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:45][D][esp32_ble_tracker:645]:   TX Power: 2
[20:08:47][D][esp32_ble_tracker:620]: Found device 53:29:EF:0D:82:3E RSSI=-55
[20:08:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:47][D][esp32_ble_tracker:620]: Found device 60:79:35:4D:3F:BD RSSI=-87
[20:08:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[20:08:47][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
```

the blue light on the desk is still flashing at this point, like it didn't successfully pair.

I tried [your recommendation](https://github.com/j5lien/esphome-idasen-desk-controller/issues/25#issuecomment-883697499) with similar results.

Can I do anything to debug this?

rcambrj, 2021-07-31:
This bug seems to affect a few users and it seems evident that nobody knows what's going on. I wonder, maybe it's something to do with the hardware we're using? ESP32 boards come in different shapes and sizes. I'm using this one: https://www.amazon.nl/dp/B071P98VTG

sliwma, 2021-07-31:
I'm using clones of Lolin32 Lite.

ju1ced, 2021-08-04:
> 
> 
> This bug seems to affect a few users and it seems evident that nobody knows what's going on. I wonder, maybe it's something to do with the hardware we're using? ESP32 boards come in different shapes and sizes. I'm using this one: https://www.amazon.nl/dp/B071P98VTG

I have the same issues and i'm using the same board you are using. The AZDelivery ESP32 NodeMCU from amazon

rcambrj, 2021-08-04:
it would be great if someone who has this working could speak up about which board they're using. I'd be happy to ship one of these AZDelivery boards to you.

angadsingh, 2021-08-09:
having the same issue as the above user and other's in the tracker.

it seems that the firmware on the latest controller have new service and characteristic UUIDs?

check this out @j5lien:

```
[11:56:30][I][ble_client:083]: Attempting BLE connection to c2:6f:1c:b7:d0:f9
[11:56:30][I][ble_sensor:033]: [Desk Height] Connected successfully!
[11:56:30][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[11:56:30][I][idasen_desk_controller:042]: [Desk] Connected successfully!
[11:56:30][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[11:56:32][I][ble_client:159]: Service UUID: 0x1800
[11:56:32][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[11:56:32][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[11:56:32][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[11:56:32][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[11:56:32][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[11:56:32][I][ble_client:159]: Service UUID: 0x1801
[11:56:32][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[11:56:32][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[11:56:32][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[11:56:32][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[11:56:32][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[11:56:32][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[11:56:32][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[11:56:32][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[11:56:32][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[11:56:32][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[11:56:32][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[11:56:32][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[11:56:32][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[11:56:32][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[11:56:32][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[11:56:32][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[11:56:32][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
```

angadsingh, 2021-08-09:
i'm able to see the desk height under this service and characteristic correctly using bluetility on the mac:
[image]

```
    service_uuid: '99FA0020-338A-1024-8A49-009C0215F78A'
    characteristic_uuid: '99FA0021-338A-1024-8A49-009C0215F78A'
```

however setting up with the same as just a desk height sensor does not work with ESPHome (no idasen desk controller involved).

the client just connects, gets the services and characteristics, and then disconnects after some time. meanwhile the desk can only move down very slowly and can't move up anymore.

i've noticed that on connecting the desk to the mac (bluetility or other apps) - the desk actually sends a pairing request. could it be the case that esphome is not accepting it's pairing request?

[image]

angadsingh, 2021-08-09:
was able to perfectly control the desk using https://github.com/rhyst/idasen-controller on my mac. so it is something do with ESPHome or ESP32 or this project

angadsingh, 2021-08-12:
here are the logs showing what happens:
```
[12:23:45][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:45][I][ble_client:083]: Attempting BLE connection to c2:6f:1c:b7:d0:f9
[12:23:46][I][ble_client:159]: Service UUID: 0x1800
[12:23:46][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x9
[12:23:46][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[12:23:46][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[12:23:46][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[12:23:46][I][ble_client:339]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[12:23:46][I][ble_client:159]: Service UUID: 0x1801
[12:23:46][I][ble_client:160]:   start_handle: 0xa  end_handle: 0xd
[12:23:46][I][ble_client:339]:  characteristic 0x2A05, handle 0xc, properties 0x20
[12:23:46][I][ble_client:159]: Service UUID: 99FA0001338A10-248A-4900-9C02-15F78A
[12:23:46][I][ble_client:160]:   start_handle: 0xe  end_handle: 0x13
[12:23:46][I][ble_client:339]:  characteristic 99FA0002338A10-248A-4900-9C02-15F78A, handle 0x10, properties 0xc
[12:23:46][I][ble_client:339]:  characteristic 99FA0003338A10-248A-4900-9C02-15F78A, handle 0x12, properties 0x12
[12:23:46][I][ble_client:159]: Service UUID: 99FA0010338A10-248A-4900-9C02-15F78A
[12:23:46][I][ble_client:160]:   start_handle: 0x14  end_handle: 0x17
[12:23:46][I][ble_client:339]:  characteristic 99FA0011338A10-248A-4900-9C02-15F78A, handle 0x16, properties 0x1e
[12:23:46][I][ble_client:159]: Service UUID: 99FA0020338A10-248A-4900-9C02-15F78A
[12:23:46][I][ble_client:160]:   start_handle: 0x18  end_handle: 0x1f
[12:23:46][I][ble_client:339]:  characteristic 99FA0021338A10-248A-4900-9C02-15F78A, handle 0x1a, properties 0x12
[12:23:46][I][ble_client:339]:  characteristic 99FA0029338A10-248A-4900-9C02-15F78A, handle 0x1d, properties 0x2
[12:23:46][I][ble_client:339]:  characteristic 99FA002A338A10-248A-4900-9C02-15F78A, handle 0x1f, properties 0x2
[12:23:46][I][ble_client:159]: Service UUID: 99FA0030338A10-248A-4900-9C02-15F78A
[12:23:46][I][ble_client:160]:   start_handle: 0x20  end_handle: 0xffff
[12:23:46][I][ble_client:339]:  characteristic 99FA0031338A10-248A-4900-9C02-15F78A, handle 0x22, properties 0xc
[12:23:46][D][esp32_ble_tracker:180]: Starting scan...
[12:23:46][D][esp32_ble_tracker:620]: Found device 03:C3:AB:3F:CA:56 RSSI=-66
[12:23:46][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:46][D][esp32_ble_tracker:620]: Found device 78:45:70:3D:91:3A RSSI=-66
[12:23:46][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:46][D][esp32_ble_tracker:645]:   TX Power: 2
[12:23:46][D][esp32_ble_tracker:620]: Found device 78:0A:13:D6:2A:F3 RSSI=-44
[12:23:46][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:47][D][esp32_ble_tracker:620]: Found device DC:A9:04:8A:18:92 RSSI=-48
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:47][D][esp32_ble_tracker:620]: Found device 35:78:E8:D9:06:A5 RSSI=-96
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:47][D][esp32_ble_tracker:620]: Found device 3C:55:0D:B8:66:D2 RSSI=-95
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:47][D][esp32_ble_tracker:620]: Found device 4C:24:B9:98:38:7B RSSI=-33
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:47][D][esp32_ble_tracker:645]:   TX Power: 2
[12:23:47][D][esp32_ble_tracker:620]: Found device 98:E0:D9:A4:C7:96 RSSI=-87
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:47][D][esp32_ble_tracker:620]: Found device C2:6F:1C:B7:D0:F9 RSSI=-37
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:47][D][esp32_ble_tracker:645]:   TX Power: 2
[12:23:47][D][esp32_ble_tracker:620]: Found device 19:85:21:B7:50:AB RSSI=-73
[12:23:47][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:48][D][esp32_ble_tracker:620]: Found device 5D:35:02:E5:7D:B5 RSSI=-45
[12:23:48][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:48][D][esp32_ble_tracker:620]: Found device 24:FC:E5:24:5A:1A RSSI=-97
[12:23:48][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:49][D][esp32_ble_tracker:620]: Found device 72:B1:59:1E:6B:DB RSSI=-89
[12:23:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:49][D][esp32_ble_tracker:620]: Found device 54:F6:D4:6B:C9:F7 RSSI=-97
[12:23:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:50][D][esp32_ble_tracker:620]: Found device 0D:BB:8F:D9:FD:C7 RSSI=-98
[12:23:50][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:50][D][esp32_ble_tracker:620]: Found device A4:30:7A:EC:4C:57 RSSI=-91
[12:23:50][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:52][D][esp32_ble_tracker:620]: Found device 40:CB:C0:D1:BC:D2 RSSI=-91
[12:23:52][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[12:23:52][D][esp32_ble_tracker:645]:   TX Power: 2
[12:23:53][D][esp32_ble_tracker:620]: Found device 44:43:6F:CB:E2:54 RSSI=-90
[12:23:53][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:23:55][D][esp32_ble_tracker:620]: Found device 49:80:A2:23:9B:BC RSSI=-96
[... 62 lines cut ...]
[12:24:49][D][esp32_ble_tracker:620]: Found device 5A:33:87:30:A5:BB RSSI=-62
[12:24:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:24:49][D][esp32_ble_tracker:620]: Found device 72:B1:59:1E:6B:DB RSSI=-91
[12:24:49][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[12:24:49][D][ble_client:045]: Found device at MAC address [C2:6F:1C:B7:D0:F9]
[... cut ...]
[12:24:49]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[12:24:49][W][ble_client:115]: connect to c2:6f:1c:b7:d0:f9 failed, status=133
[12:24:49][D][esp32_ble_tracker:180]: Starting scan...
[12:24:49][D][ble_client:045]: Found device at MAC address [C2:6F:1C:B7:D0:F9]
[12:24:49][I][ble_client:083]: Attempting BLE connection to c2:6f:1c:b7:d0:f9
[12:24:50]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
[12:24:50][W][ble_client:115]: connect to c2:6f:1c:b7:d0:f9 failed, status=133
[12:24:50][D][esp32_ble_tracker:180]: Starting scan...
```

angadsingh, 2021-08-12:
@j5lien update: works with version 1.2.0 - the one before you moved to esphome's ble_client! :)

esphome's ble_client isn't doing its job properly. please either figure out the issue there or give users option to use your component's bluetooth manager directly!

[image]

MunkeyBalls, 2021-08-14:
> 
> 
> @j5lien update: works with version 1.2.0 - the one before you moved to esphome's ble_client! :)
> 
> esphome's ble_client isn't doing its job properly. please either figure out the issue there or give users option to use your component's bluetooth manager directly!
> 
> [image]

I can confirm it works with version 1.2.0 

Unrelated question, is there a way move the desk to a certain % ?

angadsingh, 2021-08-16:
yes once you expose the desk as a cover using esphome, then position control works perfectly

- you can test it by manually using the `cover.set_cover_position` service in home assistant
- you can do it visually by adding the cover.desk entity to lovelace
- you may also use the awesome [linak desk card](https://github.com/IhorSyerkov/linak-desk-card) in HACS that was built to work with this component:

[image]
- lastly, you can expose this cover to google assistant and just say "hey google, set the position of the desk to 70%". you can even create a google routing like "raise the desk", "lower the desk", etc. and make it run this command, for your favorite positions
```
google_assistant:
  project_id: !api-project-id
  service_account: !api-project.json
  report_state: true
  entity_config:
    cover.idasen:
      expose: true
      name: Desk
      room: Master bedroom
```
- you can use the cover in HA automations. e.g. i use it to flash the phillips hue lightstrips mounted behind the desk **while the desk moves**:
```
alias: Idasen Desk Movement Effects
description: ''
trigger:
  - platform: state
    entity_id: binary_sensor.idasen_desk_moving
    to: 'on'
condition: []
action:
  - service: scene.create
    data:
      snapshot_entities:
        - light.hue_lightstrip_plus
      scene_id: lightstrip_state_before_idasen_mvmt_effect
  - service: light.turn_on
    target:
      device_id: 68927737201dbecacf069475c87514c8
  - service: light.turn_on
    target:
      device_id: 68927737201dbecacf069475c87514c8
    data:
      color_name: firebrick
      flash: long
      brightness_pct: 100
  - wait_template: '{{ states(''binary_sensor.idasen_desk_moving'') == ''off'' }}'
  - service: light.turn_off
    target:
      device_id: 68927737201dbecacf069475c87514c8
  - scene: scene.lightstrip_state_before_idasen_mvmt_effect
mode: single
```

Nicras, 2021-08-20:
can confirm it works with 1.2.0 but only for the sensors, not for the cover...when trying to move the desk from HA it doesn't

rmervine, 2021-09-02:
I’m having the same issue here. And if I’m understanding correctly, using older version of this repository at 1.2.0 is a workaround?

if I set the external component to 1.2.0, I run into syntax issues. For example Idasen Desk Controller is expecting a MAC address even though it’s defined in ble_client.  And I tried the readme from the 1.2.0 zip file of applying that code, but its complaining in ESPHome with the syntax errors. I’m having some issues posting my code as it’s not formatting correctly when I use the insert code. I’ll keep working on it.

Can someone post their 1.2.0 config here?

angadsingh, 2021-09-02:
here's the full working config with 1.2.0:

```
esphome:
  name: idasencontroller
  platform: ESP32
  board: esp32cam

logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "foobar"
  password: "barfoo"

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v1.2.0

idasen_desk_controller:
    # Desk controller bluetooth mac address
    # -----------
    # Required
    mac_address: "C2:6F:1C:B7:D0:F9"
    # Use bluetooth callback to update sensors and cover entities
    # Deactivate this option when data are not correctly updated when using the cover entity
    # -----------
    # Optionnal (default true)
    bluetooth_callback: false

cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  - platform: idasen_desk_controller
    desk_height:
      # Height in cm
      name: "Desk Height"
      # Millimeter precision
      accuracy_decimals: 1

binary_sensor:
  # Desk bluetooth connection
  - platform: idasen_desk_controller
    name: "Desk Connection"
    type: CONNECTION
  # Desk moving status
  - platform: idasen_desk_controller
    name: "Desk Moving"
    type: MOVING
```

rmervine, 2021-09-02:
@angadsingh ,

Thank you for posting your config. I was able to get this config this morning by figuring out how to get the readme webpage on 1.2.0. Your config matched what was on that page.

I can connect to it through Home Assistant. However, using the cover or the card does not result in the desk moving. If I use the card, I can see it trying move and if I use the cover, then the move command loops. 

Oh well it's progress and I will has that part out.

angadsingh, 2021-09-03:
Thats weird. Working perfectly for me from home assistant. To stop the endless loop problem - i applied this patch:
https://github.com/j5lien/esphome-idasen-desk-controller/pull/18

mentioned in this issue: https://github.com/j5lien/esphome-idasen-desk-controller/issues/17

for applying the patch you have to download this project's 1.2.0 version, apply the code changes in the above pull request and then refer to the component locally in the yaml like this:
```
external_components:
  - source: ./esphome-idasen-desk-controller-1.2.0/components
```

this is how your folder structure should look like:
[image]

rmervine, 2021-09-03:
There is something that I have which may stand out.  I do have the Linak DPG1C controller. I updated my Idasen desk with this controller.  Do you by any chance have the factory one from Ikea?  I noticed @Nicras  has the same controller I have in another post to where he is indicating issues. And since he mentioned sensors is working for him but not cover. It's possible it may not end up working with this controller. 

Quick side note, I understand what you posted about pulling locally and making the changes manually. However, one thing I don't get is how do you get the files to reference locally in ESPHome. I'm very new to ESPHome and really the only thing I got is to add a YAML file. I assume you have to upload the files to the EPSHome to a certain location.

hoborm, 2021-09-04:
> There is something that I have which may stand out.  I do have the Linak DPG1C controller. I updated my Idasen desk with this controller.  Do you by any chance have the factory one from Ikea?  I noticed @Nicras  has the same controller as me in another post to where he is indicating issues. And since he mentioned sensors is working for him but not cover in this post. It's possible it may not end up working with this controller. 
> 
> Quick side note, I understand what you posted about pulling locally and making the changes manually. However, one thing I don't get is how do you get the files to reference locally in ESPHome. I'm very new to ESPHome and really the only thing I got is to add a YAML file. I assume you have to upload the files to the EPSHome to a certain location. 

The ESPHome compiles your yaml file to the esp device. In the case of compiling source code you would have to upload the code with the CLI tool from your computer provided by the ESPHome team. Make sure you read through the documentation to understand the whole picture.

rmervine, 2021-09-10:
Thanks Hoborm. It was quite easy to do after reading the documentation from the command line.  The endless loop problem did end up getting resolved after uploading the source code changes. However, the issue of the desk not responding to the move commands from Home Assistant have remained. I have to chalk it up to my controller being a DPG1C.

sliwma, 2021-12-15:
Anyone knows how to use v1.2.0 now? I have one board that was not updated as it's working, but I need to change the SSID and add MQTT Client, and even after downloading oldest available ESPhome version I got that:
[image]

omhy, 2021-12-16:
Here is my full working config @ ESP32 Wroom

`esphome:
  name: desk-old
  platform: ESP32
  board: esp32dev
  libraries:
    - "ESP32 BLE Arduino"

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxx"

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Desk-Old Fallback Hotspot"
    password: "xxx"

captive_portal:

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v1.2.2
      
idasen_desk_controller:
    # Desk controller bluetooth mac address
    # -----------
    # Required
    mac_address: "MAC"
    # Use bluetooth callback to update sensors and cover entities
    # Deactivate this option when data are not correctly updated when using the cover entity
    # -----------
    # Optionnal (default true)
    bluetooth_callback: true

cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  - platform: idasen_desk_controller
    desk_height:
      # Height in cm
      name: "Desk Height"
      # Millimeter precision
      accuracy_decimals: 1
      
binary_sensor:
  # Desk bluetooth connection
  - platform: idasen_desk_controller
    name: "Desk Connection"
    type: CONNECTION
  # Desk moving status
  - platform: idasen_desk_controller
    name: "Desk Moving"
    type: MOVING`

fabiosoft, 2022-01-20:
using 1.2.2 for me fixed connection reboot, but it doesn't update the current position while moving but only when i force click on stop.

anyway the height is totally wrong :/

Here is the log...desk is halfway but it reads 14cm OMG! 🤣

```
[22:36:50][D][idasen_desk_controller:208]: Desk bluetooth data: height 14.4 - speed 0.0
[22:36:50][D][idasen_desk_controller:209]: Speed 0.0 Counter: 1.0
[22:36:50][D][sensor:113]: 'Desk Height': Sending state 14.40000 cm with 1 decimals of accuracy
[22:36:50][D][cover:172]: 'Desk' - Publishing:
[22:36:50][D][cover:175]:   Position: 22%
[22:36:50][D][cover:188]:   Current Operation: IDLE
```

fabiosoft, 2022-01-25:
Maybe i found a solution for updating position while moving... but it's arduino code and no ESPHome... how can i convert the code?

mensfeld, 2023-04-25:
@fabiosoft where can I find the code?