j5lien/esphome-idasen-desk-controller #33: Linak BLE2LIN support

agners, 2021-10-03:
I have a Desk with the Linak controller and a Linak BLE2LIN002 Bluetooth adapter (ver 1.08). Similar to what @lars-1 reported in https://github.com/j5lien/esphome-idasen-desk-controller/issues/7#issuecomment-886203111 the device disconnects quite quickly, and seems not to be able to read an actual height.

Is the Protocol of the Linak BLE2LIN known to be different to Ikea Idasen?

What is I find interesting is that the first time the `esp32_ble_tracker` does not list the MAC of the BLE2LIN adapter. But it finds it shortly after. In the next scan it appears, and shortly after the desk disconnects. Could it be that the scan actually leads to the disconnect?

I also don't understand where `No sensor characteristic found at service 8AF71502-9C00-498A-2410-8A332000FA99 char 8AF71502-9C00-498A-2410-8A332100FA99` is coming from. Those UUID's seem to exist nowhere, I wonder if something goes wrong during parsing/transferring UUIDs.

I am using ESPHome 2021.9.2 on a ESP32 dev board.

```
[19:52:55][D][esp32_ble_tracker:180]: Starting scan...
[19:52:55][V][component:207]: Component esp32_ble_tracker took a long time for an operation (0.06 s).
[19:52:55][V][component:208]: Components should block for at most 20-30ms.
[19:52:55][D][esp32_ble_tracker:628]: Found device 12:23:23:34:45:56 RSSI=-64
[19:52:55][D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[19:52:56][D][esp32_ble_tracker:628]: Found device 12:23:23:34:45:57 RSSI=-84
[19:52:56][D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[19:52:56][D][esp32_ble_tracker:651]:   Name: '....'
[19:52:56][V][esp32_ble_tracker:599]: Unhandled type: advType: 0x12
...
[19:53:44][D][ble_client:045]: Found device at MAC address [E4:1E:AA:AA:AA:AA]
[19:53:44][I][ble_client:083]: Attempting BLE connection to e4:1e:aa:aa:aa:aa
[19:53:44][V][ble_client:113]: [e4:1e:aa:aa:aa:aa] ESP_GATTC_OPEN_EVT
[19:53:44][I][ble_sensor:033]: [Desk Height] Connected successfully!
[19:53:44][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[19:53:44][I][idasen_desk_controller:042]: [] Connected successfully!
[19:53:44][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[19:53:49][V][ble_client:132]: cfg_mtu status 0, mtu 23
[19:53:49][V][ble_client:157]: [e4:1e:aa:aa:aa:aa] ESP_GATTC_SEARCH_CMPL_EVT
[19:53:49][I][ble_client:159]: Service UUID: 0x1800
[19:53:49][I][ble_client:160]:   start_handle: 0x1  end_handle: 0x7
[19:53:49][I][ble_client:339]:  characteristic 0x2A00, handle 0x3, properties 0xa
[19:53:49][I][ble_client:339]:  characteristic 0x2A01, handle 0x5, properties 0x2
[19:53:49][I][ble_client:339]:  characteristic 0x2A04, handle 0x7, properties 0x2
[19:53:49][I][ble_client:159]: Service UUID: 0x1801
[19:53:49][I][ble_client:160]:   start_handle: 0x8  end_handle: 0xb
[19:53:49][I][ble_client:339]:  characteristic 0x2A05, handle 0xa, properties 0x20
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0xb
[19:53:49][I][ble_client:159]: Service UUID: 99FA0001-338A-1024-8A49-009C0215F78A
[19:53:49][I][ble_client:160]:   start_handle: 0xc  end_handle: 0x11
[19:53:49][I][ble_client:339]:  characteristic 99FA0002-338A-1024-8A49-009C0215F78A, handle 0xe, properties 0xc
[19:53:49][I][ble_client:339]:  characteristic 99FA0003-338A-1024-8A49-009C0215F78A, handle 0x10, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x11
[19:53:49][I][ble_client:159]: Service UUID: 99FA0010-338A-1024-8A49-009C0215F78A
[19:53:49][I][ble_client:160]:   start_handle: 0x12  end_handle: 0x15
[19:53:49][I][ble_client:339]:  characteristic 99FA0011-338A-1024-8A49-009C0215F78A, handle 0x14, properties 0x1e
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x15
[19:53:49][I][ble_client:159]: Service UUID: 0x180A
[19:53:49][I][ble_client:160]:   start_handle: 0x16  end_handle: 0x1a
[19:53:49][I][ble_client:339]:  characteristic 0x2A29, handle 0x18, properties 0x2
[19:53:49][I][ble_client:339]:  characteristic 0x2A24, handle 0x1a, properties 0x2
[19:53:49][I][ble_client:159]: Service UUID: 99FA0020-338A-1024-8A49-009C0215F78A
[19:53:49][I][ble_client:160]:   start_handle: 0x1b  end_handle: 0x37
[19:53:49][I][ble_client:339]:  characteristic 99FA0021-338A-1024-8A49-009C0215F78A, handle 0x1d, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x1e
[19:53:49][I][ble_client:339]:  characteristic 99FA0022-338A-1024-8A49-009C0215F78A, handle 0x20, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x21
[19:53:49][I][ble_client:339]:  characteristic 99FA0023-338A-1024-8A49-009C0215F78A, handle 0x23, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x24
[19:53:49][I][ble_client:339]:  characteristic 99FA0024-338A-1024-8A49-009C0215F78A, handle 0x26, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x27
[19:53:49][I][ble_client:339]:  characteristic 99FA0025-338A-1024-8A49-009C0215F78A, handle 0x29, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x2a
[19:53:49][I][ble_client:339]:  characteristic 99FA0026-338A-1024-8A49-009C0215F78A, handle 0x2c, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x2d
[19:53:49][I][ble_client:339]:  characteristic 99FA0027-338A-1024-8A49-009C0215F78A, handle 0x2f, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x30
[19:53:49][I][ble_client:339]:  characteristic 99FA0028-338A-1024-8A49-009C0215F78A, handle 0x32, properties 0x12
[19:53:49][V][ble_client:374]:    descriptor 0x2902, handle 0x33
[... 23 lines cut ...]
[19:53:49][V][component:207]: Component esp32_ble_tracker took a long time for an operation (0.49 s).
[19:53:49][V][component:208]: Components should block for at most 20-30ms.
[19:53:49][W][ble_sensor:119]: [Desk Height] Cannot poll, not connected
[19:53:49][W][ble_sensor:119]: [Desk Speed] Cannot poll, not connected
[19:53:50][D][esp32_ble_tracker:628]: Found device E4:1E:aa:aa:aa:aa RSSI=-54
[19:53:50][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[19:53:50][D][esp32_ble_tracker:628]: Found device 12:23:23:34:45:56 RSSI=-64
...
[19:53:54][W][ble_sensor:082]: Error reading char at handle 29, status=5
[19:53:54][W][ble_sensor:082]: Error reading char at handle 29, status=5
[19:53:54][W][idasen_desk_controller:104]: Error reading char at handle 29, status=5
...
[19:54:19][V][ble_client:140]: [e4:1e:aa:aa:aa:aa] ESP_GATTC_DISCONNECT_EVT
[19:54:19][W][ble_sensor:039]: [Desk Height] Disconnected!
[19:54:19][V][sensor:070]: 'Desk Height': Received new state nan
[19:54:19][D][sensor:121]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[19:54:19][W][ble_sensor:039]: [Desk Speed] Disconnected!
[19:54:19][V][sensor:070]: 'Desk Speed': Received new state nan
[19:54:19][D][sensor:121]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
[19:54:19][W][idasen_desk_controller:049]: [] Disconnected!
[19:54:19][D][binary_sensor:036]: 'Desk Connection': Sending state OFF
[19:54:19][D][ble_client:045]: Found device at MAC address [E4:1E:aa:aa:aa:aa]
[19:54:19][I][ble_client:083]: Attempting BLE connection to e4:1e:aa:aa:aa:aa
...
```

agners, 2021-10-03:
Just realized that `8AF71502-9C00-498A-2410-8A332000FA99` is the reverse of the actual characteristic. I guess this is just a formatting error when this messages is being printed.

anisart, 2021-11-22:
@agners it's a esphome bug that was fixed in latest version https://esphome.io/changelog/2021.11.0.html#ble-sensor-uuid-changes

agners, 2021-11-28:
@anisart hm, there was also this printing error at one point: https://github.com/esphome/esphome/pull/2061, but this seems to be a more sever issue indeed.

Did you try with ESPHome 2021.11, does it work for you? I'll test it later next month, currently don't have access to the desk.

anisart, 2021-11-29:
@agners yes, I tried ble_client with ESPHome 2021.11 and it works fine without characteristic reversing