j5lien/esphome-idasen-desk-controller #36: Component idasen_desk_controller requires component esp32.

joshuajeschek, 2021-11-08:
Hey, this is my configuration:
```yaml
esphome:
  name: idasen
  platform: ESP32
  board: esp32dev

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v3.0.0

esp32_ble_tracker:
ble_client:
  - mac_address: "00:00:00:00:00:00"
    id: idasen_desk

idasen_desk_controller:
  ble_client_id: idasen_desk
  only_up_down_command: false

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "..."

wifi:
  ssid: "..."
  password: "..."
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Idasen Fallback Hotspot"
    password: "..."

captive_portal:

```

When validating the configuration, I get the following error:
```bash
INFO Reading configuration /config/esphome/idasen.yaml...
Failed config

idasen_desk_controller: [source /config/esphome/idasen.yaml:16]
  
  Component idasen_desk_controller requires component esp32.
  ble_client_id: idasen_desk
  only_up_down_command: False
```

Do you have any idea why that might be? Any help is appreciated, thanks :D

sud33p, 2021-11-14:
@joshuajeschek were you able to fix this issue? I have the exact same error, and the same config as yours which I followed from the setup instruction. Not sure where the issue is...

joshuajeschek, 2021-11-14:
It works now, but I have no idea how i fixed it! I think i reinstalled it on the esp device at some point and maybe also reloaded home assistant...
But now I have another problem, the device won't connect to my desk. I always get code 133, which really helps with debugging (not)

sud33p, 2021-11-14:
Ok, got it fixed too. Noticed there was an update to my ESPHome, updated it to 2021.10.3 and don't get the validation error anymore. But now, yes I am getting the same error as yours, code 133 when trying to connect to my desk.

joshuajeschek, 2021-11-14:
yeah, really weird. looks like it can't keep the connection up.
Have you also pressed the pairing button on the desk?`
For me, there is this error when I press the button (after it connected successfully):
`[W][idasen_desk_controller:104]: Error reading char at handle 26, status=5`
Do you get the same?

joshuajeschek, 2022-02-02:
This issue has resolved itself, but I have run into a new issue, described in #42
@sud33p Do you have the same issue as I described there?

closed 2022-02-02 by joshuajeschek