j5lien/esphome-idasen-desk-controller #37: ESP32 keeps connecting and disconnecting

Dutch-Jojo, 2021-11-24:
**Problem**: ESP32 connects and disconnects to the IDASEN ikea desk.
**System using**: ESPhome on desktop which connects to the ESPhome dashboard
**Script used**:

        esphome:
          name: desk
          platform: ESP32
          board: nodemcu-32s

        esp32_ble_tracker:
        
        globals:
          # To store the Desk Connection Status
          - id: ble_client_connected
            type: bool
            initial_value: 'false'
        
        ble_client:
          - mac_address: "EE:04:05:A7:F4:2D"
            id: idasen_desk
            on_connect:
              then:
                # Update the Desk Connection Status
                - lambda: |-
                    id(ble_client_connected) = true;
                - delay: 5s
                # Update desk height and speed sensors after bluetooth is connected
                - lambda: |-
                    id(desk_height).update();
                    id(desk_speed).update();
            on_disconnect:
              then:
                # Update the Desk Connection Status
                - lambda: |-
                    id(ble_client_connected) = false;
        
        sensor:
          # Desk Height Sensor
          - platform: ble_client
            ble_client_id: idasen_desk
            id: desk_height
            name: 'Desk Height'
            service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
            characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
            icon: 'mdi:arrow-up-down'
            unit_of_measurement: 'cm'
            accuracy_decimals: 1
            update_interval: never
            notify: true
            lambda: |-
              uint16_t raw_height = ((uint16_t)x[1] << 8) | x[0];
              unsigned short height_mm = raw_height / 10;
        
              return (float) height_mm / 10;
        
          # Desk Speed Sensor
          - platform: ble_client
            ble_client_id: idasen_desk
            id: desk_speed
            name: 'Desk Speed'
            service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
            characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
            icon: 'mdi:speedometer'
            unit_of_measurement: 'cm/min' # I'm not sure this unit is correct
            accuracy_decimals: 0
            update_interval: never
            notify: true
            lambda: |-
              uint16_t raw_speed = ((uint16_t)x[3] << 8) | x[2];
              return raw_speed / 100;
[... 170 lines cut ...]
    [16:44:59][D][sensor:113]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
    [16:44:59][W][ble_client:117]: connect to ee:04:05:a7:f4:2d failed, status=133
    [16:44:59][D][esp32_ble_tracker:214]: Starting scan...
    [16:44:59][D][esp32_ble_tracker:719]: Found device 71:E2:4B:5F:AA:32 RSSI=-65
    [16:44:59][D][esp32_ble_tracker:740]:   Address Type: RANDOM
    [16:45:00][D][ble_client:047]: Found device at MAC address [EE:04:05:A7:F4:2D]
    [16:45:00][I][ble_client:085]: Attempting BLE connection to ee:04:05:a7:f4:2d
    [16:45:00]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
    [16:45:00][W][ble_sensor:039]: [Desk Height] Disconnected!
    [16:45:00][D][sensor:113]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
    [16:45:00][W][ble_sensor:039]: [Desk Speed] Disconnected!
    [16:45:00][D][sensor:113]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
    [16:45:00][W][ble_client:117]: connect to ee:04:05:a7:f4:2d failed, status=133
    [16:45:00][D][esp32_ble_tracker:214]: Starting scan...
    [16:45:00][D][ble_client:047]: Found device at MAC address [EE:04:05:A7:F4:2D]
    [16:45:00][I][ble_client:085]: Attempting BLE connection to ee:04:05:a7:f4:2d
    [16:45:00]lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
    [16:45:00][W][ble_sensor:039]: [Desk Height] Disconnected!
    [16:45:00][D][sensor:113]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
    [16:45:00][W][ble_sensor:039]: [Desk Speed] Disconnected!
    [16:45:00][D][sensor:113]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
    [16:45:00][W][ble_client:117]: connect to ee:04:05:a7:f4:2d failed, status=133
    [16:45:00][D][esp32_ble_tracker:214]: Starting scan...

I have looked everywhere, tried other ESP32, checked if any device was still connected. But nothing. Please help.

blackcult, 2021-12-04:
Hello,
I tried to replicate your problem and got the same thing, the latest ESPhome version.

simonszu, 2021-12-27:
Hi, 
same here. 

`[09:57:02][D][ble_client:047]: Found device at MAC address [F5:4D:1B:8E:41:9B]
[09:57:02][I][ble_client:085]: Attempting BLE connection to f5:4d:1b:8e:41:9b
[09:57:03][I][idasen_desk_controller:042]: [Desk] Connected successfully!
[09:57:03][I][ble_sensor:033]: [Desk Height] Connected successfully!
[09:57:03][I][ble_sensor:033]: [Desk Speed] Connected successfully!
[09:57:03][D][binary_sensor:036]: 'Desk Connection': Sending state ON
[09:57:04][I][ble_client:161]: Service UUID: 0x1800
[09:57:04][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[09:57:04][I][ble_client:341]:  characteristic 0x2A00, handle 0x3, properties 0xa
[09:57:04][I][ble_client:341]:  characteristic 0x2A01, handle 0x5, properties 0x2
[09:57:04][I][ble_client:341]:  characteristic 0x2A04, handle 0x7, properties 0x2
[09:57:04][I][ble_client:341]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[09:57:04][I][ble_client:161]: Service UUID: 0x1801
[09:57:04][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xd
[09:57:04][I][ble_client:341]:  characteristic 0x2A05, handle 0xc, properties 0x20
[09:57:04][I][ble_client:161]: Service UUID: 99FA0001-338A-1024-8A49-009C0215F78A
[09:57:04][I][ble_client:162]:   start_handle: 0xe  end_handle: 0x13
[09:57:04][I][ble_client:341]:  characteristic 99FA0002-338A-1024-8A49-009C0215F78A, handle 0x10, properties 0xc
[09:57:04][I][ble_client:341]:  characteristic 99FA0003-338A-1024-8A49-009C0215F78A, handle 0x12, properties 0x12
[09:57:04][I][ble_client:161]: Service UUID: 99FA0010-338A-1024-8A49-009C0215F78A
[09:57:04][I][ble_client:162]:   start_handle: 0x14  end_handle: 0x17
[09:57:04][I][ble_client:341]:  characteristic 99FA0011-338A-1024-8A49-009C0215F78A, handle 0x16, properties 0x1e
[09:57:04][I][ble_client:161]: Service UUID: 99FA0020-338A-1024-8A49-009C0215F78A
[09:57:04][I][ble_client:162]:   start_handle: 0x18  end_handle: 0x1f
[09:57:04][I][ble_client:341]:  characteristic 99FA0021-338A-1024-8A49-009C0215F78A, handle 0x1a, properties 0x12
[09:57:04][I][ble_client:341]:  characteristic 99FA0029-338A-1024-8A49-009C0215F78A, handle 0x1d, properties 0x2
[09:57:04][I][ble_client:341]:  characteristic 99FA002A-338A-1024-8A49-009C0215F78A, handle 0x1f, properties 0x2
[09:57:04][I][ble_client:161]: Service UUID: 99FA0030-338A-1024-8A49-009C0215F78A
[09:57:04][I][ble_client:162]:   start_handle: 0x20  end_handle: 0xffff
[09:57:04][I][ble_client:341]:  characteristic 99FA0031-338A-1024-8A49-009C0215F78A, handle 0x22, properties 0xc
[09:57:04][W][ble_client:172]: No descriptor found for notify of handle 0x1a
[09:57:04][W][ble_client:172]: No descriptor found for notify of handle 0x1a
[09:57:04][D][esp32_ble_tracker:214]: Starting scan...
[09:57:05][D][esp32_ble_tracker:719]: Found device 2B:A2:00:F0:01:24 RSSI=-67
[09:57:05][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:05][D][esp32_ble_tracker:719]: Found device F3:33:82:8E:C9:0C RSSI=-90
[09:57:05][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:05][D][esp32_ble_tracker:742]:   Name: 'B4E5532C2EAA33AEE9'
[09:57:05][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:06][D][esp32_ble_tracker:719]: Found device E7:E7:D3:34:21:E7 RSSI=-81
[09:57:06][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:06][D][esp32_ble_tracker:742]:   Name: 'F8B3C54C704333A3E9'
[09:57:06][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:06][D][esp32_ble_tracker:719]: Found device 20:92:87:1B:39:2E RSSI=-66
[09:57:06][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:07][D][esp32_ble_tracker:719]: Found device F3:33:35:98:C3:AD RSSI=-87
[09:57:07][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:07][D][esp32_ble_tracker:742]:   Name: 'B4E553393F8030AEE9'
[09:57:07][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:08][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[09:57:08][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:08][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:08][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[09:57:08][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:08][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:09][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[09:57:09][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:09][W][ble_sensor:082]: Error reading char at handle 26, status=5
[09:57:10][D][esp32_ble_tracker:719]: Found device 68:9E:97:1E:68:23 RSSI=-66
[09:57:10][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:10][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:16][D][esp32_ble_tracker:719]: Found device 31:57:68:8E:4C:F1 RSSI=-99
[09:57:16][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:17][D][esp32_ble_tracker:719]: Found device 76:CB:9B:09:AC:40 RSSI=-65
[09:57:17][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[09:57:17][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:19][D][esp32_ble_tracker:719]: Found device 18:B9:05:C8:AE:50 RSSI=-38
[09:57:19][D][esp32_ble_tracker:740]:   Address Type: PUBLIC
[09:57:19][D][esp32_ble_tracker:742]:   Name: 'LEDnetWF010033C8AE4F'
[09:57:20][D][esp32_ble_tracker:719]: Found device 50:DE:06:75:20:37 RSSI=-67
[09:57:20][D][esp32_ble_tracker:740]:   Address Type: PUBLIC
[09:57:20][D][esp32_ble_tracker:744]:   TX Power: 2
[09:57:37][W][idasen_desk_controller:049]: [Desk] Disconnected!
[09:57:37][W][ble_sensor:039]: [Desk Height] Disconnected!
[09:57:37][D][sensor:113]: 'Desk Height': Sending state nan cm with 1 decimals of accuracy
[09:57:37][W][ble_sensor:039]: [Desk Speed] Disconnected!
[09:57:37][D][sensor:113]: 'Desk Speed': Sending state nan cm/min with 0 decimals of accuracy
[09:57:38][D][binary_sensor:036]: 'Desk Connection': Sending state OFF`

fabiosoft, 2022-01-20:
Same for me

FlatTangelo, 2022-01-21:
Same for me!

MagnusPihl, 2022-01-22:
Same problem here.

sergiomanso, 2022-01-23:
I'm having the same problem with ESPHome 2022.1.1 and esphome-idasen-desk-controller@v3.0.0
My logs report exactly the same issue as the OP

fabiosoft, 2022-01-25:
Maybe i found a solution for updating position while moving... but it's arduino code and no ESPHome... how can i convert the code?

klowdo, 2022-01-28:
Found this comment in a homeassistant thread.. this worked for me! first pair with old libs and one paired flash with new!

[Work-around](https://community.home-assistant.io/t/linak-desk-controller/292391/26)

so this is not a soulution but a work-around

FlatTangelo, 2022-01-29:
> Found this comment in a homeassistant thread.. this worked for me! first pair with old libs and one paired flash with new!
> 
> [Work-around](https://community.home-assistant.io/t/linak-desk-controller/292391/26)
> 
> so this is not a soulution but a work-around

This work-around worked for me! Thanks for the tips!

fabiosoft, 2022-02-04:
Finally! Worked for me too :)

simonszu, 2022-02-05:
Did at least someone open a bug in esphome since they need to fix that apparently?

klowdo, 2022-02-09:
but is it a esphome bug or a bug in this lib @simonszu ?

ammmze, 2022-03-21:
> but is it a esphome bug or a bug in this lib @simonszu ?

Depends on how you look at it I guess. In the 1.x series of this custom component, the custom component managed the bluetooth connection on it's own using the arduino library. Looking at the source for that, it they have implemented support for SMP (Security Manager Protocol). In the 2.x+ versions, all the manual bluetooth connection stuff was dropped in favor of the ble_client implementation in esphome. BUT the esphome implementation doesn't yet support SMP, so in the 2.x+ series, you probably cannot actually fully pair the device. So esphome basically initiates a connection, but doesn't fully establish the connection, so then it disconnects after about 30s when the desk drops out of pairing mode.

I'm in the process of trying to implement some of the SMP stuff so that this should work...though i'm not normally a C++ developer, so there's a bit of learning curve. I can verify that during the connection process, we (esphome) are receiving a `ESP_GAP_BLE_SEC_REQ_EVT` event, that the current esphome implementation does nothing with. This event is basically the desk telling the esphome that it wants security stuff set up.

So I guess, technically it's a bug that this custom component swapped the bluetooth implementation pre-maturely, given that esphome doesn't support the pairing stuff the desk is attempting to do. But the fix is ultimately that esphome needs to be updated to support the security protocol.