j5lien/esphome-idasen-desk-controller #38: 1.2.2 version problem

darkkatarsis, 2021-11-26:
Hey, are all dependencies included in the last release? I am not able to compile the application

Error:
`INFO Reading configuration /config/esphome/desk.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing desk (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/desk/.pioenvs/desk/src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp.o
In file included from src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:1:0:
src/esphome/components/idasen_desk_controller/idasen_desk_controller.h:7:23: fatal error: BLEDevice.h: No such file or directory

*******************************************************************
* Looking for BLEDevice.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:BLEDevice.h"
* Web  > https://platformio.org/lib/search?query=header:BLEDevice.h
*
*******************************************************************

compilation terminated.
*** [/data/desk/.pioenvs/desk/src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp.o] Error 1
========================== [FAILED] Took 2.66 seconds ==========================`

my conf:
`esphome:
  name: desk
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Desk Fallback Hotspot"
    password: ""

captive_portal:

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v1.2.2
      
idasen_desk_controller:
    # Desk controller bluetooth mac address
    # -----------
    # Required
    mac_address: "C8:2C:49:BC:C0:57"
    # Use bluetooth callback to update sensors and cover entities
    # Deactivate this option when data are not correctly updated when using the cover entity
    # -----------
    # Optionnal (default true)
    bluetooth_callback: false

cover:
  - platform: idasen_desk_controller
    name: "Desk"`

elmurato, 2021-12-11:
Hi @darkkatarsis 

I had the same problem and found an important information in the history of the README file. You have to add the missing library in the _esphome_ section of the configuration file. So in your case it should look like this:

```
esphome:
  name: desk
  platform: ESP32
  board: esp32dev
  libraries:
    - "ESP32 BLE Arduino"

[...]
```

I hope this helps you.

darkkatarsis, 2021-12-22:
Works now, thank you

closed 2021-12-22 by darkkatarsis