j5lien/esphome-idasen-desk-controller #41: Height/Moving sensors not update

fabiosoft, 2022-01-21:
I must use version 1.2.2 in order to avoi dconnection/disconnection loop but while moving the height sensor doesn't update unless i press the stop button for the cover.

Here is the log
```
[19:39:25][C][mdns:085]:   Hostname: deskcontrol
[19:39:25][C][idasen_desk_controller:100]: Idasen Desk Controller:
[19:39:25][C][idasen_desk_controller:101]:   Mac address: F5:45:F6:xx:xx:xx
[19:39:25][C][idasen_desk_controller:102]:   Bluetooth callback: true
[19:39:25][C][idasen_desk_controller:103]:   Desk height 'Desk Height'
[19:39:25][C][idasen_desk_controller:103]:     State Class: ''
[19:39:25][C][idasen_desk_controller:103]:     Unit of Measurement: 'cm'
[19:39:25][C][idasen_desk_controller:103]:     Accuracy Decimals: 1
[19:39:25][C][idasen_desk_controller:103]:     Icon: 'mdi:arrow-up-down'
[19:39:25][C][idasen_desk_controller:104]:   Desk moving 'Desk Moving'
[19:39:25][C][idasen_desk_controller:104]:     Device Class: 'moving'
[19:39:25][C][idasen_desk_controller:105]:   Desk connection 'Desk Connection'
[19:41:03][D][cover:078]: 'Desk' - Setting
[19:41:03][D][cover:082]:   Command: STOP
[19:41:03][D][idasen_desk_controller:329]: Cover control - STOP
[19:41:03][D][idasen_desk_controller:208]: Desk bluetooth data: height 17.1 - speed 0.0
[19:41:03][D][idasen_desk_controller:209]: Speed 0.0 Counter: 1.0
[19:41:03][D][sensor:113]: 'Desk Height': Sending state 17.10000 cm with 1 decimals of accuracy
[19:41:03][D][cover:172]: 'Desk' - Publishing:
[19:41:03][D][cover:175]:   Position: 26%
[19:41:03][D][cover:188]:   Current Operation: IDLE
```

fabiosoft, 2022-01-21:
Here is my yaml

```
esphome:
  name: deskcontrol
  libraries:
    - "ESP32 BLE Arduino"

esp32:
  board: esp32dev
  framework:
    type: arduino

logger:
api:
ota:
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap: {}
captive_portal:
time:
  - platform: homeassistant
    id: homeassistant_time

###### DESK IDASEN #######
external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v1.2.2
idasen_desk_controller:
    # Desk controller bluetooth mac address
    # -----------
    # Required
    mac_address: "f5:45:f6:3d:53:12"
    # Use bluetooth callback to update sensors and cover entities
    # Deactivate this option when data are not correctly updated when using the cover entity
    # -----------
    # Optionnal (default true)
    bluetooth_callback: true

cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  - platform: idasen_desk_controller
    desk_height:
      # Height in cm
      name: "Desk Height"
      # Millimeter precision
      accuracy_decimals: 1
      
binary_sensor:
  # Desk bluetooth connection
  - platform: idasen_desk_controller
    name: "Desk Connection"
    type: CONNECTION
  # Desk moving status
  - platform: idasen_desk_controller
    name: "Desk Moving"
    type: MOVING
```

CoMPaTech, 2022-07-01:
Thats because they only update on 'bluetooth connect' now ... if you change the `update_interval` of both sensors accordingly (from `never` to `500s` for example) you will get updates

```
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_height
    (...)
    update_interval: 500s
```

CoMPaTech, 2022-07-01:
Can't seem to get an event to trigger an automation, but having the above in at least provides some information. If you want a manual button you can use something like the below in addition to have a manual refresh, but I really think the `current_operation` data isn't moving across (I do see it in the code, but I do not get any events or state changes if I move the desk). I'm fine with lower update rates, but if you feel like hitting a button, that would work also:

```
button:
  - platform: template
    name: Update Desk Information
    id: update_desk_data
    on_press:
        # Update desk height and speed sensors on button press
        - lambda: |-
            id(desk_height).update();
            id(desk_speed).update();
```