j5lien/esphome-idasen-desk-controller #42: Error reading char at handle 26, status=5

joshuajeschek, 2022-02-02:
Hi, I'm having a weird problem, preventing me to use this at all. As far as I know, @sud33p has the same issue.

I wanted to set up an ESP32 with for my Idasen desk with this config:
```yml
esphome:
  name: idasen
  platform: ESP32
  board: esp32dev

# === DESK ===

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v3.0.0

esp32_ble_tracker:
ble_client:
  - mac_address: "CA:B3:52:CE:1E:70" # Replace with the desk bluetooth mac address
    id: idasen_desk

idasen_desk_controller:
    # Reference to the ble client component id
    # -----------
    # Required
  ble_client_id: idasen_desk
    # Fallback to use only up and down commands (less precise)
    # -----------
    # Optional
  only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "Desk"
```
However, I keep getting the same error:
```console
[16:36:03][D][esp32_ble_tracker:214]: Starting scan...
[16:36:03][D][esp32_ble_tracker:720]: Found device DE:9E:9A:10:97:6B RSSI=-86
[16:36:03][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:03][D][ble_client:047]: Found device at MAC address [CA:B3:52:CE:1E:70]
[16:36:03][I][ble_client:085]: Attempting BLE connection to ca:b3:52:ce:1e:70
[16:36:04][W][idasen_desk_controller:049]: [Desk] Disconnected!
[16:36:04][W][ble_client:117]: connect to ca:b3:52:ce:1e:70 failed, status=133
```
This error is repeating over and over again, until I press the pairing button on my desk. Then this happens:
```console
[16:36:04][D][esp32_ble_tracker:214]: Starting scan...
[16:36:04][D][ble_client:047]: Found device at MAC address [CA:B3:52:CE:1E:70]
[16:36:04][I][ble_client:085]: Attempting BLE connection to ca:b3:52:ce:1e:70
[16:36:04][I][idasen_desk_controller:042]: [Desk] Connected successfully!
[16:36:05][I][ble_client:161]: Service UUID: 0x1800
[16:36:05][I][ble_client:162]:   start_handle: 0x1  end_handle: 0x9
[16:36:05][I][ble_client:341]:  characteristic 0x2A00, handle 0x3, properties 0xa
[16:36:05][I][ble_client:341]:  characteristic 0x2A01, handle 0x5, properties 0x2
[16:36:05][I][ble_client:341]:  characteristic 0x2A04, handle 0x7, properties 0x2
[16:36:05][I][ble_client:341]:  characteristic 0x2AA6, handle 0x9, properties 0x2
[16:36:05][I][ble_client:161]: Service UUID: 0x1801
[16:36:06][I][ble_client:162]:   start_handle: 0xa  end_handle: 0xd
[16:36:06][I][ble_client:341]:  characteristic 0x2A05, handle 0xc, properties 0x20
[16:36:06][I][ble_client:161]: Service UUID: 99FA0001-338A-1024-8A49-009C0215F78A
[16:36:06][I][ble_client:162]:   start_handle: 0xe  end_handle: 0x13
[16:36:06][I][ble_client:341]:  characteristic 99FA0002-338A-1024-8A49-009C0215F78A, handle 0x10, properties 0xc
[16:36:06][I][ble_client:341]:  characteristic 99FA0003-338A-1024-8A49-009C0215F78A, handle 0x12, properties 0x12
[16:36:06][I][ble_client:161]: Service UUID: 99FA0010-338A-1024-8A49-009C0215F78A
[16:36:06][I][ble_client:162]:   start_handle: 0x14  end_handle: 0x17
[16:36:06][I][ble_client:161]: Service UUID: 99FA0020-338A-1024-8A49-009C0215F78A
[16:36:06][I][ble_client:162]:   start_handle: 0x18  end_handle: 0x1f
[16:36:06][I][ble_client:341]:  characteristic 99FA0021-338A-1024-8A49-009C0215F78A, handle 0x1a, properties 0x12
[16:36:06][I][ble_client:341]:  characteristic 99FA0029-338A-1024-8A49-009C0215F78A, handle 0x1d, properties 0x2
[16:36:06][I][ble_client:341]:  characteristic 99FA002A-338A-1024-8A49-009C0215F78A, handle 0x1f, properties 0x2
[16:36:06][I][ble_client:161]: Service UUID: 99FA0030-338A-1024-8A49-009C0215F78A
[16:36:06][I][ble_client:162]:   start_handle: 0x20  end_handle: 0xffff
[16:36:06][I][ble_client:341]:  characteristic 99FA0031-338A-1024-8A49-009C0215F78A, handle 0x22, properties 0xc
[16:36:06][D][esp32_ble_tracker:214]: Starting scan...
[16:36:06][D][esp32_ble_tracker:720]: Found device CA:B3:52:CE:1E:70 RSSI=-56
[16:36:06][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:06][D][esp32_ble_tracker:745]:   TX Power: 2
[16:36:06][D][esp32_ble_tracker:720]: Found device FE:96:45:99:85:E0 RSSI=-92
[16:36:06][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:06][D][esp32_ble_tracker:743]:   Name: 'Charge 4'
[16:36:07][D][esp32_ble_tracker:720]: Found device 37:F7:7F:51:C3:81 RSSI=-50
[16:36:07][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:10][D][esp32_ble_tracker:720]: Found device B4:E8:42:2D:D0:89 RSSI=-66
[16:36:10][D][esp32_ble_tracker:741]:   Address Type: PUBLIC
[16:36:10][D][esp32_ble_tracker:743]:   Name: 'LEDnetWF0100332DD088'
[16:36:11][W][idasen_desk_controller:104]: Error reading char at handle 26, status=5
[16:36:13][D][esp32_ble_tracker:720]: Found device DE:9E:9A:10:97:6B RSSI=-84
[16:36:13][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:22][D][esp32_ble_tracker:720]: Found device 7E:C8:A9:90:13:FC RSSI=-82
[16:36:22][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:25][D][esp32_ble_tracker:720]: Found device C4:80:7F:64:1D:5F RSSI=-90
[16:36:25][D][esp32_ble_tracker:741]:   Address Type: RANDOM
[16:36:36][W][idasen_desk_controller:049]: [Desk] Disconnected!
```
Looks promising at first, but then the error pops up, it disconnects again and the loop with the first log I showed continues.
Any idea what could be causing this?
If you need any more information, please let me know!
Thank you.

joshuajeschek, 2022-02-07:
Looking at the documentation for BLE error codes:
| Status | Status Hex | Description                    |
| ------ | ---------- | ------------------------------ |
| 5      | 0x05       | BLE_HCI_AUTHENTICATION_FAILURE |

Seems to be some sort of authentication failure. Any idea why? I pressed the pairing button, maybe I could also try completely resetting the desk

DJAMIRSAM, 2022-02-24:
hey did you get a fix for this?

joshuajeschek, 2022-02-24:
I didn't! If you have any additional information though, please let me know!

DJAMIRSAM, 2022-02-24:
I was thinking maybe the desk is asking for a code but esphome does not support codes yet

joshuajeschek, 2022-02-24:
mmmh, that could be a reason but I wouldn't think so, this is working for so many other people
what kind of board are you using?

DJAMIRSAM, 2022-02-25:
esp 32 and my table is a new idasen

joshuajeschek, 2022-02-25:
Okay, I bought mine in september 2020 🤔

nenadmilano, 2022-02-27:
Got the same errors in the log unfortunately. Was working with 1.2.3 until a few days ago but now neither 3.0.0. nur 1.2.3 is working.

DJAMIRSAM, 2022-03-09:
hmm so its nothing new then, where did u get your esp 32 i got mine off amazon it may be the ble chip (which i doubt)

DJAMIRSAM, 2022-03-09:
@j5lien can you please look into this?

andrethrill, 2022-03-17:
I have the same issue :( 

> I was thinking maybe the desk is asking for a code but esphome does not support codes yet If there was a code I’d would be 0000 or 1111

I don't think this is the case, since when pairing the desk to my phone it didn't ask for any pairing code

leukanos, 2022-03-18:
> Workaround:
> 
> 1. add v1.2.3 with the accompanying config and install
> 2. overwrite with the 3.0.0 config
> 
> Sensors are not updating, though...

In my case workaround doesn't help. Actually, it helps for a moment, and then I get the same error.

andrethrill, 2022-03-18:
> Happens to me, too.
> 
> Workaround:
> 
>     1. add v1.2.3 with the accompanying config and install
> 
>     2. overwrite with the 3.0.0 config
> 
> 
> Sensors are not updating, though...

In the end, I tried this yesterday, and it worked! I've used v1.2.2, because v1.2.3 did not work for me. Saw this workaround originally [here](https://community.home-assistant.io/t/linak-desk-controller/292391/26) and followed its steps.

It is working since yesterday, both moving the desk and reading the sensors

To do the initial pairing with the desk used the following config:

```yaml
esphome:
  name: *********    <- put the desired name here
  platform: ESP32
  board: esp32dev
  libraries:
    - "ESP32 BLE Arduino"
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ***************************************

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "*********"
    password: "**************"

captive_portal:

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v1.2.2
    

idasen_desk_controller:
  mac_address: "**:**:**:**:**:**"

  
cover:
  - platform: idasen_desk_controller
    name: "Desk"
```

Once paired successfully, used the following config:

```yaml
esphome:
  name: *********    <- put the desired name here

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ***************************************

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: *********************
    password:  *********************

captive_portal:

external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v3.0.0
    
esp32_ble_tracker:

globals:
  # To store the Desk Connection Status
  - id: ble_client_connected
    type: bool
    initial_value: 'false'

ble_client:
  - mac_address: "**:**:**:**:**:**"
    id: idasen_desk
    on_connect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = true;
        - delay: 5s
        # Update desk height and speed sensors after bluetooth is connected
        - lambda: |-
            id(desk_height).update();
            id(desk_speed).update();
    on_disconnect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = false;

idasen_desk_controller:
  ble_client_id: idasen_desk
  only_up_down_command: false
  
cover:
  - platform: idasen_desk_controller
    name: "Desk"

sensor:
  # Desk Height Sensor
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_height
    name: 'Desk Height'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:arrow-up-down'
    unit_of_measurement: 'cm'
    accuracy_decimals: 1
    update_interval: never
    notify: true
    lambda: |-
      uint16_t raw_height = ((uint16_t)x[1] << 8) | x[0];
      unsigned short height_mm = raw_height / 10;

      return (float) height_mm / 10;

  # Desk Speed Sensor
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_speed
    name: 'Desk Speed'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:speedometer'
    unit_of_measurement: 'cm/min' # I'm not sure this unit is correct
    accuracy_decimals: 0
    update_interval: never
    notify: true
    lambda: |-
      uint16_t raw_speed = ((uint16_t)x[3] << 8) | x[2];
      return raw_speed / 100;

binary_sensor:
  # Desk Bluetooth Connection Status
  - platform: template
    name: 'Desk Connection'
    id: desk_connection
    lambda: 'return id(ble_client_connected);'

  # Desk Moving Status
  - platform: template
    name: 'Desk Moving'
    id: desk_moving
    lambda: 'return id(desk_speed).state > 0;'

```

leukanos, 2022-03-18:
Thanks for sharing. I'll try this approach.

leukanos, 2022-03-18:
Reading sensors works, for now, however moving is not :(
I'll see if it keep working over longer period of time

leukanos, 2022-03-21:
And it got disconnected after 2 days and doesn't connect anymore...

joshuajeschek, 2022-03-21:
@leukanos at least you progressed 3 handles! 😁
does anybody know if it is possible to use esphome at the pull requests head?

zmasek, 2022-03-21:
I did pretty much what [andrethrill](https://github.com/andrethrill) said in the comment below me. Things work, but the sensors don't :/ I'm using Adafruit ESP32 feather board if that's any help. The controls don't work for down (up and stop do), but the automation does (because it's behaving like a curtain).

d0ugal, 2022-04-15:
I’m also seeing this issue. I just got a new desk, partly because I wanted to automate it with this project 😄

Is there anything I can do to test/debug? It isn’t clear from reading the thread if there are any leads to follow up on.

Otherwise I’ll do my own investigation

DJAMIRSAM, 2022-04-15:
Unfortunately we have not found a fix yet. But please send your progress here so others don’t have the same issue

joshuajeschek, 2022-04-15:
small update from me:
- I've updated my esphome in home assistant to [this version](https://github.com/esphome/esphome/releases/tag/2022.4.0b2) (look at these release notes, they changed the install method: https://esphome.io/changelog/2022.2.0.html)

- it worked for a bit, but the [card](https://github.com/IhorSyerkov/linak-desk-card) showed wrong values so i tinkered a bit with the configuration. The reported height is correct now, but the connection status is false and sometimes it randomly stops working - I haven't found out why yet.

- I'll post my configuration here later, probably tomorrow. I'll also continue to look into why the bugs are still occuring

ammmze, 2022-04-16:
See below for what I'm running currently. This is running on ESPHome 2022.3.2. I have 2 `external_components` setup:

1. My fork of this repository, which I've enhanced (or fixed?) the ability to auto drive on the desks that don't have auto drive. For example, when the component received a request to go to a certain position, it was sending that to the desk and the desk would start moving, but it would only move a couple of inches at most and then stop. But the component didn't attempt to continue sending the command even though it could see that it wasn't at the right position. I'm not sure if this was intentional or not...but regardless, I can auto drive my desk now. I still need to figure out the calibration though because I'm pretty sure the numbers are off, but I haven't measured. Also there are some edge cases where I've found that it gets stuck in the loop continuously sending the command to move, but the desk doesn't move. However because the ESP keeps sending the command, the paddle on the desk becomes non-responsive until I restart the ESP (and maybe the desk). Also, I think when the desk is moving, there are a _lot_ of updates that come through, but I don't think all of them make it to home assistant. The buffer of queued messages gets too full. So for example, on my home assistant card I have a stand position at 117cm and if I press it, the desk does it's thing and it does in fact make it to the position, but home assistant may get the last update at like 115cm. A quick tap of the paddle that effectively doesn't move the desk much causes one more update to run through and home assistant shows it at 117cm.
2. The PR I had submitted to ESPHome, which I believe is in the 2022.4.0b1 release

```yaml
esphome:
  name: office-desk-controller-frontier

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Office-Desk-Controller-Frontier"
    password: !secret ap_password

captive_portal:

external_components:
  - source: github://ammmze/esphome-idasen-desk-controller@make-it-work
  - source: github://pr#3320
    components:
      # list all components modified by this Pull Request here
      - ble_client
      - esp32_ble_tracker

globals:
  # To store the Desk Connection Status
  - id: ble_client_connected
    type: bool
    initial_value: 'false'

esp32_ble_tracker:

ble_client:
  - mac_address: "00:00:00:00:00:00" # Replace with the desk bluetooth mac address
    id: idasen_desk
    on_connect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = true;
        - delay: 5s
        # Update desk height and speed sensors after bluetooth is connected
        - lambda: |-
            id(desk_height).update();
            id(desk_speed).update();
    on_disconnect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = false;

idasen_desk_controller:
    # Reference to the ble client component id
    # -----------
    # Required
    ble_client_id: idasen_desk
    # Fallback to use only up and down commands (less precise)
    # -----------
    # Optional
    only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "Office Desk"

sensor:
  # Desk Height Sensor
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_height
    name: 'Office Desk Height'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:arrow-up-down'
    unit_of_measurement: 'cm'
    accuracy_decimals: 1
    update_interval: never
    notify: true
    # filters:
    #   - debounce: 0.5s
    lambda: |-
      uint16_t raw_height = ((uint16_t)x[1] << 8) | x[0];
      unsigned short height_mm = raw_height / 10;

      return (float) height_mm / 10;

  # Desk Speed Sensor
  - platform: ble_client
    ble_client_id: idasen_desk
    id: desk_speed
    name: 'Office Desk Speed'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:speedometer'
    unit_of_measurement: 'cm/min' # I'm not sure this unit is correct
    accuracy_decimals: 0
    update_interval: never
    notify: true
    # filters:
    #   - debounce: 0.5s
    lambda: |-
      uint16_t raw_speed = ((uint16_t)x[3] << 8) | x[2];
      return raw_speed / 100;

binary_sensor:
  # Desk Bluetooth Connection Status
  - platform: template
    name: 'Office Desk Connection'
    id: desk_connection
    lambda: 'return id(ble_client_connected);'

  # Desk Moving Status
  - platform: template
    name: 'Office Desk Moving'
    id: desk_moving
    lambda: 'return id(desk_speed).state > 0;'

```

DJAMIRSAM, 2022-04-18:
well thanks to you this is the first time HA moved my desk

d0ugal, 2022-04-18:
@ammmze Neat! I also tried this config out and it seems to partially work. I can now move the desk up and down with Home Assistant. However, the sensors don't seem to be updating. I don't need them for now, it isn't quite as nice but I can work with a cover percentage in Home Assistant so I just removed them for now. My more minimal config therefore look like...

```
external_components:
  - source: github://ammmze/esphome-idasen-desk-controller@make-it-work
  - source: github://pr#3320
    components:
      - ble_client
      - esp32_ble_tracker

substitutions:
   nicename: Study
   devicename: study

esphome:
  name: ${devicename}
  platform: ESP32
  board: esp32dev

# Enable logging
logger:
  level: INFO

# Enable Home Assistant API
api:

ota:
  password: "***************************************"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  
esp32_ble_tracker:
ble_client:
  - mac_address: "00:00:00:00:00:00" # Replace with the desk bluetooth mac address
    id: idasen_desk

idasen_desk_controller:
    ble_client_id: idasen_desk
    only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "Study Desk"
```

DJAMIRSAM, 2022-04-18:
@d0ugal the first time I loaded it in the sensors did not update but then I unplugged it and plugged it back in (the esp) and it’s working

d0ugal, 2022-04-18:
Good point, the height did seem to update once after restart. It didn't seem to update after that but I'll do some experimentation.

callumgare, 2023-02-27:
@d0ugal Did you ever get a chance to do some experimenting? I'm getting the same thing where the sensors are updated when the ESP first boots up but never again.

d0ugal, 2023-02-27:
@callumgare I had some problems until [4.0.0](github://j5lien/esphome-idasen-desk-controller@v4.0.0) was released. Since then it has worked fine and I don't get the issue anymore. My full config is now the following;

```
external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v4.0.0

substitutions:
   name: study-desk

esphome:
  name: ${name}
  platform: ESP32
  board: esp32dev

esp32_ble_tracker:

ble_client:
  - mac_address: "00:00:00:00:00:00" # Replace with the desk bluetooth mac address
    id: idasen_desk

idasen_desk_controller:
    ble_client_id: idasen_desk
    only_up_down_command: false

cover:
  - platform: idasen_desk_controller
    name: "Study Desk"

# These just include some common things like wifi details that I share between devices. Nothing speicifc to the desk.
<<: !include .common.yaml
<<: !include .atom-identify.yaml
```

callumgare, 2023-02-27:
Thanks for that @d0ugal!