j5lien/esphome-idasen-desk-controller #49: Are you open to being sponsord?

hco, 2022-08-10:
Hey Julien,
a friend & me have desks that could be compatible to your ESP-Home-Component, but it doesn't work reliably. 
Would you be willing to work on this for a bit of money? :) 

Our controllers are probably called dpg1m30.

arough007, 2022-08-12:
Hey,
I'm the friend.

<details>
<summary>Here is the config</summary>

```
esphome:
  name: desk

esp32:
  board: esp32dev
  framework:
    type: arduino
    
external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v3.0.0

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxx"

ota:
  password: "xxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  manual_ip:
    static_ip: 192.168.2.216
    gateway: 192.168.2.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Desk Fallback Hotspot"
    password: "wKcXidfbjOkM"

captive_portal:

esp32_ble_tracker:

ble_client:
  - mac_address: "d2:f5:86:aa:72:25" # Replace with the desk bluetooth mac address
    id: desk_home
    
idasen_desk_controller:
    # Reference to the ble client component id
    # -----------
    # Required
    ble_client_id: desk_home
    only_up_down_command: false
    
cover:
  - platform: idasen_desk_controller
    name: "Desk" 
```

</details>

<details>
<summary>And here is the log</summary>

```
INFO Reading configuration /config/tisch-max-wz.yaml...
[... 174 lines cut ...]
[20:09:53][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:53][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:53][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:54][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:54][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:54][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:54][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:54][D][esp32_ble_tracker:726]: Found device 32:36:26:E4:FB:89 RSSI=-86
[20:09:54][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[20:09:55][D][idasen_desk_controller:217]: Update Desk - Move from 36 to 97
[20:09:55][D][cover:170]: 'Desk' - Publishing:
[20:09:55][D][cover:173]:   Position: 36%
[20:09:55][D][cover:186]:   Current Operation: CLOSING
[20:09:55][D][idasen_desk_controller:200]: Update Desk - target reached
[20:09:55][D][esp32_ble_tracker:726]: Found device 46:5E:63:55:71:E6 RSSI=-84
[20:09:55][D][esp32_ble_tracker:747]:   Address Type: RANDOM
[20:09:55][D][cover:170]: 'Desk' - Publishing:
[20:09:55][D][cover:173]:   Position: 36%
[20:09:55][D][cover:186]:   Current Operation: CLOSING
[20:09:55][D][cover:170]: 'Desk' - Publishing:
[20:09:55][D][cover:173]:   Position: 36%
[20:09:55][D][cover:186]:   Current Operation: IDLE

```
</details>

j5lien, 2022-11-06:
Thank you for the offer but at the moment I don't have time to work on this project. Sorry