j5lien/esphome-idasen-desk-controller #53: Error when installing component.

Schloerhome, 2022-10-11:
Hi,  Been using this for over a year now and wanted to reinstall it ontop a new ESP32 but i keep getting error when ESPhome tries to pull the installation. 

```
Compiling /data/livingroom-esp32/.pioenvs/livingroom-esp32/src/esphome/components/sensor/filter.cpp.o
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp: In member function 'virtual void esphome::idasen_desk_controller::IdasenDeskControllerComponent::gattc_event_handler(esp_gattc_cb_event_t, esp_gatt_if_t, esp_ble_gattc_cb_param_t*)':
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:68:61: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
           esp_ble_gattc_register_for_notify(this->parent()->gattc_if, this->parent()->remote_bda, this->output_handle_);
                                                             ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:68:87: error: 'class esphome::ble_client::BLEClient' has no member named 'remote_bda'
           esp_ble_gattc_register_for_notify(this->parent()->gattc_if, this->parent()->remote_bda, this->output_handle_);
                                                                                       ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:101:50: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
       if (param->read.conn_id != this->parent()->conn_id)
                                                  ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:115:52: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
       if (param->notify.conn_id != this->parent()->conn_id || param->notify.handle != this->output_handle_)
                                                    ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp: In member function 'void esphome::idasen_desk_controller::IdasenDeskControllerComponent::write_value_(uint16_t, short unsigned int)':
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:141:65: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
   esp_err_t status = ::esp_ble_gattc_write_char(this->parent()->gattc_if, this->parent()->conn_id, handle, 2, data,
                                                                 ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:141:91: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
   esp_err_t status = ::esp_ble_gattc_write_char(this->parent()->gattc_if, this->parent()->conn_id, handle, 2, data,
                                                                                           ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp: In member function 'void esphome::idasen_desk_controller::IdasenDeskControllerComponent::read_value_(uint16_t)':
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:152:47: error: 'class esphome::ble_client::BLEClient' has no member named 'gattc_if'
       esp_ble_gattc_read_char(this->parent()->gattc_if, this->parent()->conn_id, handle, ESP_GATT_AUTH_REQ_NONE);
                                               ^
src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp:152:73: error: 'class esphome::ble_client::BLEClient' has no member named 'conn_id'
       esp_ble_gattc_read_char(this->parent()->gattc_if, this->parent()->conn_id, handle, ESP_GATT_AUTH_REQ_NONE);
                                                                         ^
*** [/data/livingroom-esp32/.pioenvs/livingroom-esp32/src/esphome/components/idasen_desk_controller/idasen_desk_controller.cpp.o] Error 1
```

V0v1kkk, 2022-10-13:
The same problem with the ESPHome 2022.9.4.

V0v1kkk, 2022-10-13:
[That PR](https://github.com/j5lien/esphome-idasen-desk-controller/pull/51) looks like a fix of the problem.

V0v1kkk, 2022-10-14:
@j5lien could you merge the PR please?

V0v1kkk, 2022-10-15:
A temporary solution:
```
external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@pull/51/head
```

j5lien, 2022-11-06:
The fix has been merged!

closed 2022-11-06 by j5lien