j5lien/esphome-idasen-desk-controller #59: Down Button not working

saruye, 2022-11-17:
First of all thanks for this awsome integration :) I have it integrated with home assistant and would have never gotten this to work on my own. 
Mostly everything is working. I can set it to a height and then the desk is moving up and down. And if I manually use the up button it's also working. But the down button is not. 
What can I do here? What information can i provide to help or where look myself?

MathijsG, 2022-12-13:
It's working fine here!

What if you try this config? (Of course change it to your credentials on some places):

esphome:
  name: idasen
  platform: esp32
  board: esp32dev # Replace with your board
  # Starting with ESPHome 2021.10, libraries need to be included manually
  libraries:
    - "ESP32 BLE Arduino"
 
# Enable logging
logger:
 
# Enable Home Assistant API
api:
 
ota:
 
wifi:
  ssid: !secret IoT_ssid
  password: !secret IoT_password
 
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Idasen Fallback Hotspot"
    password: aaaaaaaa
 
captive_portal:
 
## Begin van Ikea-specifieke dingen
external_components:
  - source: github://j5lien/esphome-idasen-desk-controller@v4.0.0
 
cover:
- platform: idasen_desk_controller
  name: "Desk"
 
idasen_desk_controller:
  # Reference to the ble client component id
  # -----------
  # Required
  ble_client_id: idasen_desk
  # Fallback to use only up and down commands (less precise)
  # -----------
  # Optional
  only_up_down_command: false
 
esp32_ble_tracker:
 
globals:
  # To store the Desk Connection Status
  - id: ble_client_connected
    type: bool
    initial_value: 'false'
 
ble_client:
  - mac_address: "DC:AC:28:84:21:E5"
    id: idasen_desk
    on_connect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = true;
        - delay: 5s
        # Update desk height and speed sensors after bluetooth is connected
        - lambda: |-
            id(desk_height).update();
            id(desk_speed).update();
    on_disconnect:
      then:
        # Update the Desk Connection Status
        - lambda: |-
            id(ble_client_connected) = false;
 
sensor:
  # Desk Height Sensor
  - platform: ble_client
    type: characteristic
    ble_client_id: idasen_desk
    id: desk_height
    name: 'Desk Height'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:arrow-up-down'
    unit_of_measurement: 'cm'
    accuracy_decimals: 1
    update_interval: never
    notify: true
    lambda: |-
      uint16_t raw_height = ((uint16_t)x[1] << 8) | x[0];
      unsigned short height_mm = raw_height / 10;
 
      return (float) height_mm / 10;
 
  # Desk Speed Sensor
  - platform: ble_client
    type: characteristic
    ble_client_id: idasen_desk
    id: desk_speed
    name: 'Desk Speed'
    service_uuid: '99fa0020-338a-1024-8a49-009c0215f78a'
    characteristic_uuid: '99fa0021-338a-1024-8a49-009c0215f78a'
    icon: 'mdi:speedometer'
    unit_of_measurement: 'cm/min' # I'm not sure this unit is correct
    accuracy_decimals: 0
    update_interval: never
    notify: true
    lambda: |-
      uint16_t raw_speed = ((uint16_t)x[3] << 8) | x[2];
      return raw_speed / 100;
 
binary_sensor:
  # Desk Bluetooth Connection Status
  - platform: template
    name: 'Desk Connection'
    id: desk_connection
    lambda: 'return id(ble_client_connected);'
 
  # Desk Moving Status
  - platform: template
    name: 'Desk Moving'
    id: desk_moving
    lambda: 'return id(desk_speed).state > 0;'

saruye, 2022-12-15:
@MathijsG Thanks for the suggestion! 
At first I tried with quite an old version which produced the issue I reported here. Then I updated and used a similar config as described here. This gave me a crash during the start which I couldn't figure out. 
After you posted this I looked at it again and realised I used a wrong board. :/ It's an esp32 but the correct board would have been nodemcu-32s. After I updated it, the crash is gone and it's working with pretty much your configuration. 
So everything good now :) 
Closing this ticket.

closed 2022-12-15 by saruye