j5lien/esphome-idasen-desk-controller #62: unable to move after a while

lukasdietz, 2023-01-06:
I am using an Linak DPG1C Controller with my desk which worked fine at first. After some time the controls got unresponsive but all sensor values appeared correctly. If i move the desk manually, all values change. 

These are my ESPHome Logs while trying to move the desk:
```
[11:35:52][D][idasen_desk_controller:217]: Update Desk - Move from 69 to 14
[11:35:53][D][idasen_desk_controller:217]: Update Desk - Move from 69 to 14
[11:35:53][D][idasen_desk_controller:217]: Update Desk - Move from 69 to 14
```
This happens until I stop pressing the button or move the desk by hand.

Is this a known issue? Could someone help me solving the problem?

p4block, 2023-02-05:
I found that pairing to the official app and back to the ESP makes it work again, but it's extremely annoying to do. For some reason my desk is not factory resetting properly, I need to toy around with that tomorrow. I think it may be related to some settings in the Linak app.

mensfeld, 2023-04-25:
@p4block any findings on that? same isue

p4block, 2023-04-25:
@mensfeld Using the official app at least once makes the integration work. The thing is, I power cycle the desk every night so repairing every day is unacceptable.

mensfeld, 2023-04-25:
@p4block thanks, I for now just move by 1 cm each 2h which seems to mitigate this.